IRI-MobileRobotics/wolf #16: Local Parameterization in Wolf

andreucm, 2016-02-17:
Create LocalParametrizationBase. Get inspirartion from [here http://ceres-solver.org/nnls_modeling.html#_CPPv2N5ceres21LocalParameterizationE] but using Eigen::Map's as in/out arguments.
Create LocalParametrizationQuaternion and Homogeneous inheriting from LocalParametrizationBase
In ceres_wrapper folder, create wrappers to each of the above class, inheriting from [ceres::LocalParameterization http://ceres-solver.org/nnls_modeling.html#_CPPv2N5ceres21LocalParameterizationE]

joansola, 2016-02-18:
Creada la classe base.

joansola, 2016-02-19:
Created LocalParametrizationQuaternion.

See if we work with Map or with pointers or with vectors.

joansola, 2016-02-27:
Created LocalParametrizationHomogeneous.

joanvallve, 2016-03-15:
The ceres local parametrization wrapper for local parametrizations should be implemented in order to use the wolf local parametrizations

joansola, 2016-03-15:
@joanvallve are you going to do the local param in the wrapper? :-D

joanvallve, 2016-03-17:
Done and pushed. Also added to CeresManager::addStateBlock(). Needs to be tested in a ceres optimization.

closed 2016-05-10 by joansola