IRI-MobileRobotics/wolf #19: Check std::vector<Fixed Size Eigen Matrix>

andreucm, 2016-02-17:
Check std::vector<Fixed Size Eigen Matrix> in all the code and add aligned allocator (second template argument of std::vector)

See http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html

joansola, 2016-05-10:
What about this issue? Can we close it?

joanvallve, 2016-05-10:
No, please, it is a TODO: check if there's a vector of fixed size eigen matrix somewhere in the wolf code...

joansola, 2016-06-17:
I found the macro

````
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION
````

in Eigen/StdVector.h, which does the trick. We can put a lot of definitions in wolf.h or in a convenient header, as so:

````
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Quaternion<double>)

EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,1,1,RowMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,2,2,RowMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,3,3,RowMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,4,4,RowMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,5,5,RowMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,6,6,RowMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,7,7,RowMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,8,8,RowMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,9,9,RowMajor>)

EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,1,1,ColMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,2,2,ColMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,3,3,ColMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,4,4,ColMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,5,5,ColMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,6,6,ColMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,7,7,ColMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,8,8,ColMajor>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,9,9,ColMajor>)

EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,2,1>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,3,1>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,4,1>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,5,1>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,6,1>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,7,1>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,8,1>)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Matrix<double,9,1>)
````

putting 'Scalar' instead of 'double', and after that, we are free to just declare:

````
vector<Vector3s> toto;
````

so we do not need to go and search all Wolf for the appearances of vector<Eigen::Stuff>