IRI-MobileRobotics/wolf #21: Processors, Trackers

joansola, 2016-02-28:
See what to do:
 - ProcessorBase uses extractFeatures() and establishConstraints()
 - ProcessorTracker uses track()
 - ProcessorMotion uses integrate()

They have similar meanings but... we need to clarify all this.

joansola, 2016-03-02:
The design of ProcessorBase is probably not going to last, at least as a base class for the others.

joansola, 2016-03-02:
We need a clear policy for managing the frames that are not yet keyframes.
- Are Frames in the Trajectory:
  - Is there just one frame
  - Or are there two Frames, last and incoming?
- Are Frames in the Tracker?

We need also a clear policy of how to transition from Tracked Frames to KeyFrames.
- Who can create key-frames? Who can crate non-key Frames?

joansola, 2016-03-03:
TODOs:
- How do we create new Keyframes (in WolfProblem ? )
- How do we create new Landmarks (in WolfProblem ?)
- How to deal with autonomous / slave Trackers ?

joanvallve, 2016-03-11:
ProcessorMotion:

After creating a KeyFrame, transfer the motion buffer as features (Delta_X) in the capture and methods interpolate(capture, timestamp) and concatenate(capture,capture) should be in the processor too.

joansola, 2016-03-17:
This issue keeps alive in issue #31 .  Closing.

closed 2016-03-17 by joansola