IRI-MobileRobotics/wolf #22: Error when compiling Wolf in a "fresh" Machine

andreucm, 2016-03-04:
Attached below, I list the error when trying to just clone and make. It is an issue with one example main, which requires qr_solver.h, and the later requires constraint_gps_pseudorange.h, which I think is deprecated. 

A fast "fix" could be to disable the compilation of this example in the examples/CMakeLists.txt

Scanning dependencies of target test_iQR_wolf2
[ 86%] Building CXX object src/examples/CMakeFiles/test_iQR_wolf2.dir/solver/test_iQR_wolf2.cpp.o
In file included from /home/acoromin/dev/wolf/src/examples/solver/test_iQR_wolf2.cpp:26:0:
/home/acoromin/dev/wolf/src/./solver/qr_solver.h:20:43: fatal error: ../constraint_gps_pseudorange.h: No such file or directory
 #include "../constraint_gps_pseudorange.h"
                                           ^
compilation terminated.
make[2]: *** [src/examples/CMakeFiles/test_iQR_wolf2.dir/solver/test_iQR_wolf2.cpp.o] Error 1
make[1]: *** [src/examples/CMakeFiles/test_iQR_wolf2.dir/all] Error 2
make: *** [all] Error 2

pt07, 2016-03-04:
yes, now the file is called constraint_gps_pseudorange2D.h
there was an include in solver/qr_solver that wasn't updated.

I'm committing the new version, tell me if there are new problems

2016-03-04 13:10 GMT+01:00 Andreu Corominas Murtra <notifications@github.com

joanvallve, 2016-03-04:
I removed the deprecated file from /usr/local/include and it compiles

andreucm, 2016-03-04:
Fixed! Thanks!

closed 2016-03-04 by andreucm