IRI-MobileRobotics/wolf #24: absolute constraints to sensor parameters

joanvallve, 2016-03-08:
Add the option of applying absolute constraints to sensor parameters (intrinsic and extrinsic). A calibration is a measure of the parameter with a covariance.

A solution may be add a list of constraints to the sensor and all methods related.

joansola, 2016-03-17:
'A solution may be add a list of constraints to the sensor and all methods related.'

The problem is that ConstraintBase points up to Feature, not Sensor. But still we can derive ConstraintBase to add an up pointer to SensorBase and do the job.

joansola, 2016-03-17:
Actually this seems really the easiest way to go. Although it is a little patchwork...

joanvallve, 2016-03-17:
I think we should discuss this... I think all stateBlocks can have an absolute constraint:

- Landmark state: As the result of other map sessions or whatever, a landmark can have an absolute constraint.
- Sensor parameter: As a result of a calibration...
- Frame: already has absolute constraints.

I think there are stateBlocks only on those 3 cases... Maybe there should be 3 kind of absolute constraints (Frame, Sensor and Landmark)