IRI-MobileRobotics/wolf #25: Unused classes?

joansola, 2016-03-11:
Are there any unused classes in WOLF? Please respond here with classes you may have written in the past but are no longer used:

 - ProcessorMotion

JaimeTarraso, 2016-03-11:
I don't know if this is what you mean, but I did create some classes (derived from others) that are not currently in use. I will use them later on, however. Should I mention them here just in case?

pt07, 2016-03-11:
`ConstraintGPSPseudorange3D` is a unused class.
I've implemented it but then, when I started trying to fuse gps and odometry, I needed something in 2D.
I was told to leave that class there, given that is implemented, because can be useful in the future to fuse with some sensor in 3D.

warning: is untested!

joansola, 2016-03-11:
I mean, classes that are not used because they were tests. They may not work, or be partial designs. In this case, we need to delete them.

If they are not used but they are functional, then we keep then.

I would prefer if you mention them here so that I know that I should not delete them.

JaimeTarraso, 2016-03-11:
Right now, I can find these:

- ConstraintPoint2D
- LandmarkPoint2D
- ProcessorImagePointBrisk

I am aware that some of them need to be renamed. In any case, the first two will be used later on, adequately modified. The last one, the ProcessorImagePointBrisk doesn't have any purpose right now, but I would want to keep it a couple of days more just in case. After that I will remove it myself. I just need to "git rm" and take it out from the CMakeLists, right?

joansola, 2016-03-11:
Jaime: for early and experimental developments, it is good if you create you own branch in git, so that the master branch is kept clean. I'll explain to you when we have some time.

closed 2016-03-14 by joansola