IRI-MobileRobotics/wolf #27: check Order of transformation at line 190 of file constraint_gps_pseudorange_2D.h

andreucm, 2016-03-13:
Line 190 is :100 :    
`Eigen::Matrix<T, 4, 4> T_map2ecef = T_tran * T_map2eno * T_enu2aux * T_lat2lon * T_lon2ecef ;`

I don't see that this was the correct order .... I would try with: 
`Eigen::Matrix<T, 4, 4> T_map2ecef = T_tran * T_lon2ecef * T_lat2lon * T_enu2aux * T_map2eno ;`

(small detail : rename T_map2eno to T_map2enu ! )

joansola, 2016-03-13:
Hello! This seems like the typical problem of frame transforms! May I suggest... What about renaming all transformations so that the meaning of T_a_b is always "transformation from b to a", which means that points in a and b are related by:

p_a = T_a_b * P_b. <-- we transformed from b to a

To make it easier to interpret, T_a_b also means "the pose of b expressed in frame a". This is the typical way we use to specify frame transforms. So there is nothing new, just the names.

It seems that you use T_map2ecef... in this case, there are 2 possibilities for interpreting this name:
1. if you mean that it transforms points in ecef to points in map, this would be T_map2ecef=T_map_ecef. And p_map = T_map_ecef * p_ecef. 
2. If it means that it transforms points in map to ecef, then it would be T_map2ecef=T_ecef_map.  And p_ecef = T_ecef_map * p_map.

Using this convention, the transformation chain involving multiple frames is easy and you cannot make mistakes, e.g.:
T_a_e = T_a_b * T_b_c * T_c_d * T_d_e

where 'b' chains with 'b', 'c' with 'c', etc, to make a total chain from 'e' to 'a' as the result states.

If the order of one of the specifications does not match, we can always do T_b_a = T_a_b^-1

It your example, the chain does not seem to follow any logical order... 

T_map2ecef = T_tran * T_map2eno * T_enu2aux * T_lat2lon * T_lon2ecef 

so it is difficult to say if it is right or wrong... only the 'enu' and 'lon' tags are properly chained, and the 'ecef' at the end seems good too, because it matches the result. 'aux', 'lat', 'tran' and 'map' do not match. And 'map' should be in the first place, instead of 'tran', because the result T_map2ecef uses 'map'

andreucm, 2016-03-13:
Fully agree Joan. We have to be extremely tidy on naming such matrixes !!

pt07, 2016-03-13:
thanks for the advice!
I've accidentally committed this version, but anyway the local copy on the IRI's pc is like this:
`T_map2ecef = T_tran * T_map2enu * T_enu2lat * T_lat2lon * T_lon2ecef ;`

The meaning i gave to this name is your second interpretation. So `sensor_p_ecef =  T_map2ecef * sensor_p_map` means 

> sensor wrt ecef = transformation of points in map to point in ecef * sensor wrt map.

There is another mistake, corrected in the local code. a the beginning we passed from enu to aux, and then from aux to lat. Then i moved aux2lat and i forgot to rename the previous matrix. It is no more like this in the new code.

I got ispiration for this naming from wolf_ros, but i'll gladly change it to the one suggested by Joan because it's easier to understand a chain.

T_tran is only a translation and i didn't know how to call it, but i'll add the aux frame in that part, given that now in no more used.

andreucm, 2016-03-13:
The translation is also a transformation between two frames (a particular case with no rotation), so in terms of naming should follow the same criteria.

joansola, 2016-03-13:
@ACM +1

I think it is generally easy to understand each individual transform. Then, you understand it --> you name it with the convention. After that, you will never again have doubts and you'll build all your chains correctly.

pt07, 2016-03-13:
yes yes, i'll add aux frame in the middle. anyway now i cannot compile for some errors in processor_brisk.h

I think that there are some private pointers used, but i want to finish writing a section of the thesis so monday morning if is still not resolved i'll take a look. Because, especially after changing name to variables, i want to compile and test before committing, that is easy to forget some.
I'll open a separate issue for that

joansola, 2016-03-13:
Oh... processor_brisk may be wrong because of me. Remove it from CMakeLists to keep going. tomorrow I'll see with Jaime how to fix this.

joansola, 2016-03-13:
Also, can you provide the error message? I could fix it now. But I cannot compile from home because I do not have a good opencv version. i switch to email for this issue

pt07, 2016-03-13:
ok was something easy. It was trying to access incoming_ptr and last_ptr, that are private member of processor_brisk. i've just added a call to the getter

pt07, 2016-03-13:
solved in commit a7a93fdbae10ed3e38a1c43f00e2201866f44ee8

closed 2016-03-13 by pt07