IRI-MobileRobotics/wolf #29: no matching function for call to ‘ProcessorBase::ProcessorBase()’

pt07, 2016-03-14:
problem compiling:

```
[ 68%] Building CXX object src/CMakeFiles/wolf.dir/processor_image_point_brisk.cpp.o
In file included from /home/ptirindelli/git/wolf/src/feature_point_image.cpp:2:0:
/home/ptirindelli/git/wolf/src/feature_point_image.h: In constructor ‘FeaturePointImage::FeaturePointImage(const Vector2s&, cv::KeyPoint, const std::vector<float>&)’:
/home/ptirindelli/git/wolf/src/feature_point_image.h:24:22: warning: ‘FeaturePointImage::keypoint_’ will be initialized after [-Wreorder]
         cv::KeyPoint keypoint_;
                      ^
[ 69%] /home/ptirindelli/git/wolf/src/feature_point_image.h:23:28: warning:   ‘std::vector<float> FeaturePointImage::descriptor_’ [-Wreorder]
         std::vector<float> descriptor_;
                            ^
/home/ptirindelli/git/wolf/src/feature_point_image.cpp:16:1: warning:   when initialized here [-Wreorder]
 FeaturePointImage::FeaturePointImage(const Eigen::Vector2s & _measurement, const cv::KeyPoint _keypoint ,const std::vector<float> & _descriptor):
 ^
Building CXX object src/CMakeFiles/wolf.dir/processor_brisk.cpp.o
/home/ptirindelli/git/wolf/src/processor_image_point_brisk.cpp: In constructor ‘ProcessorImagePointBrisk::ProcessorImagePointBrisk(int, int, float)’:
/home/ptirindelli/git/wolf/src/processor_image_point_brisk.cpp:13:9: error: class ‘ProcessorImagePointBrisk’ does not have any field named ‘ProcessorTracker’
         ProcessorTracker(PRC_TRACKER_BRISK), sensor_cam_ptr_(nullptr), capture_img_ptr_(nullptr), brisk_(_threshold, _octaves, _pattern_scales) //initializes the algoritm
         ^
/home/ptirindelli/git/wolf/src/processor_image_point_brisk.cpp:13:143: error: no matching function for call to ‘ProcessorBase::ProcessorBase()’
         ProcessorTracker(PRC_TRACKER_BRISK), sensor_cam_ptr_(nullptr), capture_img_ptr_(nullptr), brisk_(_threshold, _octaves, _pattern_scales) //initializes the algoritm
                                                                                                                                               ^
/home/ptirindelli/git/wolf/src/processor_image_point_brisk.cpp:13:143: note: candidates are:
In file included from /home/ptirindelli/git/wolf/src/node_linked.h:153:0,
                 from /home/ptirindelli/git/wolf/src/sensor_base.h:11,
                 from /home/ptirindelli/git/wolf/src/sensor_camera.h:5,
                 from /home/ptirindelli/git/wolf/src/processor_image_point_brisk.h:5,
                 from /home/ptirindelli/git/wolf/src/processor_image_point_brisk.cpp:3:
/home/ptirindelli/git/wolf/src/processor_base.h:18:9: note: ProcessorBase::ProcessorBase(ProcessorType)
         ProcessorBase(ProcessorType _tp);
         ^
/home/ptirindelli/git/wolf/src/processor_base.h:18:9: note:   candidate expects 1 argument, 0 provided
/home/ptirindelli/git/wolf/src/processor_base.h:15:7: note: ProcessorBase::ProcessorBase(const ProcessorBase&)
 class ProcessorBase : public NodeLinked<SensorBase, NodeTerminus>
       ^
/home/ptirindelli/git/wolf/src/processor_base.h:15:7: note:   candidate expects 1 argument, 0 provided
[ 70%] make[2]: *** [src/CMakeFiles/wolf.dir/processor_image_point_brisk.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
Building CXX object src/CMakeFiles/wolf.dir/active_search.cpp.o
/home/ptirindelli/git/wolf/src/processor_brisk.cpp: In member function ‘virtual ConstraintBase* ProcessorBrisk::createConstraint(FeatureBase*, LandmarkBase*)’:
/home/ptirindelli/git/wolf/src/processor_brisk.cpp:225:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
make[1]: *** [src/CMakeFiles/wolf.dir/all] Error 2
make: *** [all] Error 2
ptirindelli@ant2:~/git/wolf/build$ 
```

joansola, 2016-03-14:
OK I look at it now

pt07, 2016-03-14:
no rush for me, i have commented the line on cmakelists

joansola, 2016-03-14:
It should be fixed, but I cannot check it. Can you? If not, please re-post new error msg here.

pt07, 2016-03-14:
```
[ 95%] Built target test_state_quaternion
Linking CXX executable ../../../bin/test_wolf_tree
/home/ptirindelli/git/wolf/src/examples/test_image.cpp: In function ‘int main(int, char**)’:
/home/ptirindelli/git/wolf/src/examples/test_image.cpp:93:59: error: cannot allocate an object of abstract type ‘ProcessorBrisk’
     ProcessorBrisk* p_brisk = new ProcessorBrisk(30,0,0.5f);
                                                           ^
In file included from /home/ptirindelli/git/wolf/src/examples/test_image.cpp:8:0:
/home/ptirindelli/git/wolf/src/./processor_brisk.h:16:7: note:   because the following virtual functions are pure within ‘ProcessorBrisk’:
 class ProcessorBrisk : public ProcessorTracker
       ^
In file included from /home/ptirindelli/git/wolf/src/./processor_brisk.h:5:0,
                 from /home/ptirindelli/git/wolf/src/examples/test_image.cpp:8:
/home/ptirindelli/git/wolf/src/./processor_tracker.h:139:33: note: 	virtual ConstraintBase* ProcessorTracker::createConstraint(FeatureBase*, NodeBase*)
         virtual ConstraintBase* createConstraint(FeatureBase* _feature_ptr, NodeBase* _node_ptr) = 0;
                                 ^
make[2]: *** [src/examples/CMakeFiles/test_image.dir/test_image.cpp.o] Error 1
make[1]: *** [src/examples/CMakeFiles/test_image.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 97%] Built target test_wolf_tree
[ 98%] Built target test_wolf_prunning
[100%] Built target test_wolf_imported_graph
make: *** [all] Error 2
ptirindelli@ant2:~/git/wolf/build$ 

```

joansola, 2016-03-14:
Coming to iri. Will check there

joansola, 2016-03-14:
Fixed. Please git pull on 'master'

closed 2016-03-14 by joansola