IRI-MobileRobotics/wolf #31: ProcessorMotion

joansola, 2016-03-15:
See if we can use the prototype of ProcessorMotion to derive motion processors for:
 - IMU
 - Odometry

@AtDinesh look at it when you have a moment. Thank you.

AtDinesh, 2016-03-15:
I'll create a ProcessorPreintegratedIMU deriving from ProcessorMotion and I'll have a look on how to do it.

AtDinesh, 2016-03-15:
I'm not sure what the following functions are supposed to do exactly : 
- data2dx
- computeAverageDT
- getDt

joansola, 2016-03-15:
A brief explanation, general examples, and suggestions for your IMU 
implementation:

*data2dx* : convert the motion data to a minimal form.
example:
data is position + quaternion (7D)
dx is position + rotation vector (6D)
then, we need to convert the quaternion to angle/axis, and then to 
rotation vector.
IMU example:
data: acc, gyro
dx: the part you need to integrate into the delta_integrals (possibly, 
acc*dt, gyro*dt, but not sure)

*computeAverageDT*: computes the average dt by doing: dt = (t_end - 
t_init) / buffer.size()
example: none
IMU: I think this is already done no? You do not need to do it

*getDt*: get the processor dt.
example: it is normally fixed, so this is just a getter for dt_.
IMU example: just a getter for dt_.

Please keep updated with the branch motion. I am now commiting a test 
where the following classes are defined and tested:

test: src/examples/test_motion.cpp
classes: CaptureOdom3D, ProcessorOdom3D : they will later on be moved to 
the Wolf library.
main() : creates a sensor, with a capture with data, and the processor, 
and tests it.

It is a little buggy, but you can get inspired with ProcessorOdom3D to 
build your ProcessorIMU. Especifically, look at plus() and minus(), and 
extractData().

We'll be refining these designs. Any suggestions are welcome.

Joan

joansola, 2016-03-15:
OK there is a little bug in the test, but it compiles. You can check the 
code of the different functions. The bug I'll fix it easy and soon this 
same afternoon.

One other thing: I see you had conflicts when rebasing the motion 
branch. Apart from the conflicts, we should not be rebasing branches 
that are shared in the remote repo. There is a lot o finfo about this on 
the internet. When sharing branches among users, one should only do merge.

If you want to keep you branch apart from master, but with all of the 
changes in master up to date, you do:

git checkout motion
git merge master

This way, master is not taking the changes in motion, but motion takes 
the changes in master.

I think you know these things, but anyway...

Joan

joansola, 2016-03-15:
I pushed a new design for ProcessorMotion, called ProcessorMotion2.

It seems to me that this one is more rational and logical than the other one.

By now, only the design in the .h is done -- just the API.

@AtDinesh please have a look. We may want to discard the original ProcessorMotion -- it is very complicated.

AtDinesh, 2016-03-16:
Thank you for the explanations.

And sorry for the rebase thing ! I just saw examples on the net I understand now.

I think I'd rather use processorMotion2 than processorMotion. Do you want me to delete processorMotion ?

joansola, 2016-03-16:
Hello  Dinesh, I hope you are reading this.

We need to collaborate quite a lot during the construction of the Processor IMU. Let me put here some guidelines.

1. ProcessorMotion will implement most of the algorithmic aspects of such a Motion processor: 
  * init(), 
  * reset(), 
  * process(), 
  * advance(), 
  * makeKeyFrame(), 
  and the accessors 
  * state(), 
  * deltaState(), and 
  * sumDeltas().

This means that you do not need to re-declare these functions in your derived class.

2. The math behind these operations needs to be done in the derived class. We need to implement three distinct mathematical operations:
  * xPluxDelta() performs the sum of a state plus a delta-state
  * deltaPlusDelta() creates a delta-state that is the sum of a delta-state and another delta-state
  * xMinusX() creates a delta-state that is the substraction of two states.

This means that you must re-declare and define these functions in your derived class.

This API will be later on enriched with extra functionality for the sake of dealing with two necessary features of the class:
  * Compute all relevant Jacobians
  * Compute covariances matrix by properly integrating noise

joansola, 2016-03-16:
With regard to the IMU, we will be working in 3D and a lot of the math would be interesting to have it built elsewhere, in a kind of library, especially for what regards 3D orientations, quaternions and rotation matrices. 

In RTSLAM, I created the quatTools.hpp file, which is highly templatized and has all quaternion-related functions, with Jacobians. You can check it out [here](https://github.com/damarquezg/rtslam/blob/master/include/rtslam/quatTools.hpp). We need to come up with something similar, and this file could be a nice starting point. However, we want to take profit of Eigen, which already implements most of the quaternion functionality -- although it does not work with Jacobians.

This we need to discuss.

joansola, 2016-03-16:
Do you want me to delete processorMotion ?

No, not yet. Just leave it aside, I have some code there that I regard.

Just derive from ProcessorMotion2. We'll change names later.

joansola, 2016-03-16:
Please do very frequent git pulls and pushes! I am editing ProcessorMotion2 right now and this will affect you. If you push/pull frequently the probability of complicated merges is low.

AtDinesh, 2016-03-24:
I am working in parallel on the quatTools file from rtslam (I may implement some of the maths we will need here.)
rtslam was using jmath et ublas functions : shall I replace all of them with eigen functions or functions I will define myself ?

joansola, 2016-03-24:
Beware that Eigen already has a lot of the quaternion algebra implemented.

You can also have a look at the ProcessorOdometry3D class, where I implemented xPlusDelta(), deltaPlusDelta() and deltaMinusDelta(). You'll see most of the quaternion operations needed.

joansola, 2016-03-24:
Also, Dinesh, I would like you to determine whether these three functions xPlusDelta(), deltaPlusDelta() and deltaMinusDelta() are well adapted for your IMU problem. If they are not then maybe I will need to think the ProcessorMotion2 class over.

In particular, I want to know if you can integrate an IMU measurement into a Delta-integrated value by doing the sequence:

delta = data2delta(data);
delta_integrated = deltaPlusDelta(delta)

(the API in the class is different but the functionality is this one).

If this scheme is not good for you, then please suggest something more adapted.

AtDinesh, 2016-03-29:
For xPlusDelta() and deltaPlusDelta I think it's in the philosophy of the paper and should be OK.
I still need to think about deltatMinusDelta().

concerning format I am evaluating several options. One of them would be to create new data classes for preintegrated measurements and code rotations as matrix in deltas. I think this is also what they are doing I'm checking this. But I could also make computations as matrices and store values as vectors I expect to have a better idea at the end of the day.
But if I do this, should I not have, for example for deltas : 
deltas of position (3-vector) + orientation vector (3D-space --> 9-vector), velocity (3-vector) + biases ?

joansola, 2016-03-29:
Dont worry by now about deltaMinusDelta

And orientation error is a 3-vector. You can obtain a rot matrix from it (9 params) or a quaternion (4 params), but the delta must be expressed in minimal space, that is 3 params

joansola, 2016-03-29:
Hi Dinesh,

I pushed a version of the ProcessorMotion2 and CaptureMotion2 that use a template to define the Delta type. Please read the documentation of such classes and see if this suits better your needs (I hope so).

AtDinesh, 2016-03-30:
I can see that motion branch has been deleted from the remote repository.
Do you want me to work directly on the master branch ?
I was adding parts of the mathematics needed for IMU preintegrated data on a local branch created from motion.

joansola, 2016-03-30:
Can You keep working on your branch that departed from motion? 
Joan Solà - IRI-UPC-CSIC - (sent from cellphone)

AtDinesh, 2016-03-30:
Yes I just need to update my motion branch then.
by doing:
git checkout motion
git merge master

?

joansola, 2016-03-30:
I am not 100% sure about rebasing, but it should work

joansola, 2016-03-31:
Hi Dinesh, I think I got it. Please check out ProcessorOdometry3D. Check the Odo3dDeltaType, and the code in xPlusDelta, deltaPlusDelta and deltaMinusDelta. They should be very clear now.

To be able to work like this with structs of Delta values, I needed to add the pure virtual deltaZero() in ProcessorMotion2, that you will have to implement too in your ProcessorPreintegratedIMU. It is just a way to set the zero-delta. With the documentation and the example in ProcessorOdom3D you should quickly see what needs to be done.

See that the Deltas are structs, but the state vector is not.

Possibly, we will have to add a last function, Eigen::VectorXs deltaVector(_delta), returning a full Eigen::VectorXs for the delta, for the sake of compatibility with other parts of Wolf. We'll see this later.

joansola, 2016-04-05:
Hello @AtDinesh 

I have updated the 'Motion in Wolf' document from Google Drive. Please have a look at the last section where I explain the ProcessorMotion class.