IRI-MobileRobotics/wolf #33: StateBlock.type = eular angle. is it possible?

pt07, 2016-03-21:
In the wiki it is written that a state block can be of the type euler angle.
https://github.com/IRI-MobileRobotics/wolf/wiki/State-block

Is it still possible to set that property or is the wiki out-of-date? I cannot find it in the code.
(i'm asking only because can be useful to me, not because I want you to update the wiki if is not true)

joansola, 2016-03-21:
It is possible. However, I prefer quaternions for 3D orientation, and I would like that all methods involving 3D orientations use quaternions.

If you want to use Euler, then you will need to code all the math associated to them. 

If you want to use Quaternion, then we'll use Eigen quaternion for most of the algebra, and will build what we need extra on top of it, in an effort shared by more members of the project.

I will be out this week but I can keep discussing the topic if you wish. Otherwise we can discussit next week.

andreucm, 2016-03-21:
Hi, 
Pier , can you say more precisely to which state block you would like to use Euler ?
I think it is not necessary for us, since there are no 3D rotations involved in the state (only in the ECEF-sensor transforms).
best, 
andreu

pt07, 2016-03-21:
No I'm sorry, it isn't a 3D rotation. I misunderstood the meaning of Euler
angles.

In my case it is only one dimension, more specifically is the parameter
map_o. Is the orientation of the x-axis of the frame map with respect to
the geographical east.
It is a simple angle in radiants.. Now it converges in values outside [ 0 ;
2*M_PI]

It is not a problem, when I want to print it I only normalize it. I wanted
to know if there were a special type for that angles, or if it is ok to use
a normal state block.

2016-03-21 23:07 GMT+01:00 Andreu Corominas Murtra <notifications@github.com

joansola, 2016-03-21:
Just use one state block with a vector of 1 dimension storing this angle.

If you want to get rid of the issue 0-2pi, then we can define a local parametrization for it which does the trick for -pi : pi. It's easy.

pt07, 2016-03-21:
Ok it is how i'm doing now.
Next week I'll see if i'll have the time for seeing how the local parametrization works.
Anyway it isn't important because it works fine aswell.

Thanks for the answers, I'm going to close the issue

closed 2016-03-21 by pt07

joanvallve, 2016-03-23:
I think that there is no reason for doing a local parametrization apart
from the (-pi, pi] thing... the addition of angles does not have a special
implementation... Actually, the only thing important is keeping difference
of angles (not angles) between (-pi, pi] in the residual computation.
That's not what local parametrization does...

*Joan Vallvé Navarro*
+34 934010863 | www.iri.upc.edu/people/jvallve
Institut de Robòtica i Informàtica Industrial
C/ Llorens i Artigas 4-6, pl.-2, 08028 Barcelona, Spain

joansola, 2016-03-23:
@joanvallve  you are right. 

However, the ±pi trick can be done in the plus() function of the local parametrization, with the Jacobians as Identities. I know it is an overkill solution, but has the benefit of doing the job without requiring any other piece of code in Wolf to ensure the ±pi: at each update of the solver, plus() is called on each StateBlock having a LocalParametrization, and therefore the ±pi trick is done automatically.

It is not a big deal I think. Do as you feel.

joanvallve, 2016-03-23:
If we want to keep all angles in (-pi, pi] (only if we want to) we can do
it with a local parametrization, for sure. We can do it and see the extra
computational cost that it represents in ceres.

I insist that the important think to be guaranteed is fitting all angle
errors in (-pi, pi]. It means, in all constraints that use angle errors,
they have to be in (-pi, pi] before multiplying them by the square root of
covariance matrix. That wouldn't be solved by the local parametrization.

*Joan Vallvé Navarro*
+34 934010863 | www.iri.upc.edu/people/jvallve
Institut de Robòtica i Informàtica Industrial
C/ Llorens i Artigas 4-6, pl.-2, 08028 Barcelona, Spain