IRI-MobileRobotics/wolf #34: building wolf with catkin build

artivis, 2016-03-29:
This morning I got wolf to compile on my laptop using [catkin build](http://catkin-tools.readthedocs.org/en/latest/verbs/catkin_build.html).

This tool makes package finding easier (among other thing) and allows one to have catkin and plain cmake packages in the same workspace and build them all as long as they have a `package.xml` file that declares it as a package to be build (along with its dependencies).

Doing so I could setup and build wolf and ceres in less than a minute on my desktop.

note : I disabled gflag and glog for ceres. See [here](https://github.com/artivis/ceres_solver)

For a brief guide see : [here](https://gist.github.com/artivis/219ae6f7977d88253ed0)

We should consider to have a fork of ceres such as mine and add the package.xml files as it still allows one to compile the old fashion way while giving another (simple) option -> `catkin build`

joansola, 2016-04-08:
I am commenting just for completeness.

We discussed this issue live @artivis , @joanvallve and @joansola. 

@artivis will write new documentaiton for installing WOLF with CERES using catkin_build. For this, we'll have a fork of ceres_solver in our repos that is conveniently patched for this to work seamlessly, and also to avoid different wolf collaborators to install different versions of ceres.

We'll be able to continue working with the current cmake approach, to migrate to catkin_build, or even to have both systems coexist. So really the change affects mainly the newcomers who will have an easier life installing the packages.

artivis, 2016-08-10:
@joansola @joanvallve up.

I have updated the setup procedure that allows to build everything with `catkin build`.
It implies some additions & modifications to the wolf CMakeLists which are in overall beneficial I think.

For wolf, please check the branch 'catkin_build' 95de292567dd3fdbd7f783069fdb1ff851a4ce23.

Could someone please create a fork of Ceres (either on github or gitlab) ?
We may want to ensure a 'frozen' version of Ceres that we know works with the current wolf master. And I could set it up to also work within `catkin build`.