IRI-MobileRobotics/wolf #36: idea super_landmarks (containers)

joanvallve, 2016-04-05:
superLandmarks contain a set of landmarks whose stateBlocks pointers point to superLandmarks's ones. The residual of the landmark corner should be reimplemented accepting fixed parameters for the new frame origin (container frame instead of corner frame).

Constraints and residuals should be updated (pointers to the state change).

Just to remember the idea when we discuss it in the future.

joanvallve, 2016-04-07:
superLandmarks could be also groups of landmarks that share any geometric restriction (for instance, coplanar 3D landmarks, learned objects, etc..).

artivis, 2016-04-07:
upvote.

joanvallve, 2016-04-26:
It could be implemented using something similar to LocalParametrization...

joansola, 2016-08-02:
OK it seems that we can close this issue as @joanvallve already implemented ideas that are similar, and we are therefore not going to pursue on this any time soon.
If the necessity comes back then wel'll reponen a new issue with new motivations.

closed 2016-08-02 by joansola