IRI-MobileRobotics/wolf #37: ProcessorTracker::init()

joanvallve, 2016-04-06:
Shouldn't this function call detectNewFeatures()?

joansola, 2016-04-06:
Uhm maybe yes... it makes sense!

joanvallve, 2016-04-07:
Should a processor be initialized? maybe it should auto-initialize itself when a keyframe is created, we should implement this keyFrameCreation callback from the outside

joansola, 2016-04-07:
On keyframe creation, the tracker should be reset, not initialized. It's not exactly the same.

joanvallve, 2016-04-08:
I'll rephrase:

> Should a processor be initialized? maybe it should auto-initialize itself when **the first** keyframe is created, we should implement this keyFrameCreation callback from the outside

joansola, 2016-04-10:
I fixed many things in init(), and now we are detecting new Features and adding them on origin_ptr.

Waiting to see how it works before closing.

closed 2016-04-11 by joansola