IRI-MobileRobotics/wolf #38: Namespaces

joansola, 2016-04-06:
Create namespaces: wolf, hardware, trajectory, map.

joanvallve, 2016-04-07:
I think we should start with wolf:: and see if we need more specific namespaces

artivis, 2016-04-07:
The very least is wolf:: .
Then for inner namespace they have to be few and as meaningful as possible.
Hardware sounds good, but trajectory may not be very obvious regarding all it is supposed to contains.

joansola, 2016-04-07:
Guys, I am having real trouble only with the wolf namespace. 

I did the following:

for each file in wolf/src do{
  add after all #includes: " <code>namespace wolf { </code>"
  add at end of file (before the #endif in header files): " <code>} // namespace wolf </code>"
}

then, I got millions of compiling errors, starting with (you can checkout the branch 'namespace' and see for yourselves):

<code>
% make -j2 all 
Scanning dependencies of target wolf
[  2%] Building CXX object src/CMakeFiles/wolf.dir/capture_base.cpp.o
[  2%] Building CXX object src/CMakeFiles/wolf.dir/capture_gps_fix.cpp.o
In file included from /Users/jsola/dev/wolf/src/node_linked.h:156:0,
                 from /Users/jsola/dev/wolf/src/capture_base.h:12,
                 from /Users/jsola/dev/wolf/src/capture_base.cpp:1:
/Users/jsola/dev/wolf/src/processor_base.h: In member function 'virtual bool wolf::wolf::ProcessorBase::permittedKeyFrame()':
/Users/jsola/dev/wolf/src/processor_base.h:59:28: error: invalid use of incomplete type 'class wolf::WolfProblem'
     return getWolfProblem()->permitKeyFrame(this);
                            ^
In file included from /Users/jsola/dev/wolf/src/capture_base.h:10:0,
                 from /Users/jsola/dev/wolf/src/capture_base.cpp:1:
/Users/jsola/dev/wolf/src/wolf.h:260:7: error: forward declaration of 'class wolf::WolfProblem'
 class WolfProblem;
       ^
/Users/jsola/dev/wolf/src/capture_base.cpp: In constructor 'wolf::CaptureBase::CaptureBase(const wolf::TimeStamp&, wolf::SensorBase*)':
/Users/jsola/dev/wolf/src/capture_base.cpp:13:27: error: invalid use of incomplete type 'class wolf::SensorBase'
  sensor_p_ptr_(sensor_ptr_->getPPtr()),
                           ^
In file included from /Users/jsola/dev/wolf/src/capture_base.h:10:0,
                 from /Users/jsola/dev/wolf/src/capture_base.cpp:1:
/Users/jsola/dev/wolf/src/wolf.h:272:7: error: forward declaration of 'class wolf::SensorBase'
</code>

If you observe well the error log, sometimes you see WolfProblem, sometimes wolf::WolfProblem, and sometimes even wolf::wolf::WolfProblem. It should always be wolf::WolfProblem!! Then, something is odd, but I cannot see what!

I also added a second improvement: put all forward declarations after the "<code>namespace wolf { </code>". I also got millions of errors.

andreucm, 2016-04-07:
From googleing "namespace and forward declaration C++" It seems that when using namespaces and fwd declarations, the later change the syntaxis, and they have to be coded as:

namespace ns1
{
    class a;
}

instead of just what there was:

    class a;

or just  what intuition would indicate:

   class ns1::a;

References:
http://www.epstudiossoftware.com/?p=63
http://www.devx.com/tips/Tip/32118

joansola, 2016-04-07:
I'll try this, thanks

joansola, 2016-04-07:
Still not working. Here the first error:

/Users/jsola/dev/wolf/src/hardware_base.h: In member function 'void wolf::wolf::HardwareBase::removeSensor(const SensorBaseIter&)':
/Users/jsola/dev/wolf/src/hardware_base.h:41:32: error: no matching function for call to 'wolf::wolf::HardwareBase::removeDownNode(const SensorBaseIter&)'
     removeDownNode(_sensor_iter);
                                ^
/Users/jsola/dev/wolf/src/hardware_base.h:41:32: note: candidates are:
In file included from /Users/jsola/dev/wolf/src/capture_base.h:12:0,
                 from /Users/jsola/dev/wolf/src/capture_gps_fix.h:5,
                 from /Users/jsola/dev/wolf/src/capture_gps_fix.cpp:1:
/Users/jsola/dev/wolf/src/node_linked.h:120:14: note: void wolf::NodeLinked<UpperType, LowerType>::removeDownNode(const LowerNodeIter&) [with UpperType = wolf::wolf::WolfProblem; LowerType = wolf::wolf::SensorBase; wolf::NodeLinked<UpperType, LowerType>::LowerNodeIter = std::_List_iterator<wolf::wolf::SensorBase*>]

I see an accumulation of wolf::wolf::className. This may indicate that NodeLinked, being a recursive class, is messing up and appending more and more wolf:: namespaces as the recursivity advances.

Or so to speak...

joansola, 2016-04-07:
I like the wolves appearing in these messages :-D

joansola, 2016-04-08:
OK I think I got it. 

Ina few .h files, after the class declaration, there are some #includes before the inline definitions.

These includes need to go out of the namespace, so wrapping the whole file around namespace wolf { .. } was bad.

I am now leaving these includes out of the namespace and things start to compile. I'll post a new comment once I've finished.

andreucm, 2016-04-08:
That sounds reasonable!

joansola, 2016-04-08:
OK so I think I got it. Things that I learned:
  - <b>forward declarations inside namespace:</b> namespace wolf { class Forwarded; }
  - <b>#includes out of namespace:</b> #include "a.h" namespace wolf { ... } #include "b.h" namespace wolf { ... }
  - <b>static or global functions and variables:</b> namespace wolf { int a = 1; int b(){}; }. In a few test files, there are some static variables and functions being defined. These are OK to go inside the namespace. they could also be declared outside, with a <code>using namespace wolf;</code> inside the function, but the first option is preferable.
  - <b>main() function out of namespace but using it:</b> int main() { using namespace wolf; [...] }. The main function needs to be out of the namespace otherwise it becomes wolf::main and the system cannot find it to make the executable.

Now I need to go to IRI and finish the job, because at home I cannot compile everything. I'll keep you posted!

andreucm, 2016-04-08:
Good summary!
Regarding the main(), I've usually seen "using namespace" outside the main() , just before entering to it. 
However I'm more fan of explicitly put namespaces at objects declarations and function calls, so do not use "using namespace", and code such as:
     std::vector my_vec; 
     wolf::FrameBase my_frame; 
     std::cout << "bla, bla..." << std::endl;

joansola, 2016-04-08:
@andreucm , two things:

Putting the 'using namespace wolf' inside main() means that it only affects the main(). Not a big deal in this case, because main() is the last fcn of the test.cpp file, and the test.cpp file is not included by anyone. So putting it outsie would have affected exactly the same scope!

Regarding the 'using' versus the specification wolf::stuff, yes, you are right. For the tests, and for the sake of a painful-less migration to the wolf namespace, I opted for the using namespace wolf option: it is safe because of point above, and it requires less coding!.

joansola, 2016-04-08:
Done 'add namespace wolf to project'
Pushed to master.
Thanks to all.
Closing.

closed 2016-04-08 by joansola