IRI-MobileRobotics/wolf #39: ProcessorTracker::track() correspondences not returned

joanvallve, 2016-04-07:
`track(const FeatureBaseList& _feature_list_in, FeatureBaseList& _feature_list_out)`

This function only returns via  `_feature_list_out` the features of `_feature_list_in` found in `origin_ptr_` but the correspondences are not returned.

They should be returned somehow since it is work already done and it is necessary information, am I right?

Should we add a new output parameter, generate constraints from the features in `_feature_list_out`...? I think the first option is better.

joansola, 2016-04-07:
Yes, track() should also take care to create the constraints and append them to each Feature. 

However, mind that the constraints already exist in last_ptr, so in fact we only need to move them to origin_ptr, and not create new ones. 

Moreover, we need to destruct the Features and the Constraints in last_ptr that were not found in incoming_ptr. This destruction might be triggered by the destruction of last_ptr, but I am not sure.

@joanvallve Can you check line 122 of ProcessorTracker.h? There is a commented line destructing last_ptr, <code>//    last_ptr_->destruct();</code>. If we uncomment it, we have runtime errors when the main() finishes, possibly some pointer points to garbage...

joanvallve, 2016-04-08:
In the documentation, correction/checking the correspondence with the `origin` features and the constraints move is said to be done in `processKnown()`, that's why I ask for the correspondences to be providen..

About the error, can you specify which test should I run?

joanvallve, 2016-04-08:
In `track()` the constraints shouldn't be moved since it is called from `ProcessKnown()` and `ProcessNew()`

In `ProcessKnown()` there should be done:
1. Track the features in `last_ptr_` that are also in `incoming_ptr_`
2. Correct & check the correspondences resulting from the tracking with `origin_ptr_` features.
3. Move constraints from `last_ptr_` features to tracked `incoming_ptr_`features.
4. Destruct of non-tracked `last_ptr_` features constraints.

In `ProcessNew()` there should be done:
1. Detect new features in `last_ptr_`
2. Track the new features in `last_ptr_` that are also in `incoming_ptr_`
3. Create constraints from `last_ptr_` new features to `incoming_ptr_`features.
4. Append `last_ptr_` and `incoming_ptr_` new features to their featureList.

Since in `ProcessKnown()` and `ProcessNew()`, different things are done with the correspondences (moving the existing ones in the first case, creating new ones in the second), `track()` should provide the correspondences outside.

I am implementing `ProcessKnown()` and `ProcessNew()` in ProcessorTrackerFeature/ProcessorTrackerLandmark. I inform you when it works.

About the error, before running anything, I see an bug that may cause this error. Appending the lists is only adding down nodes to the capture but it doesn't link the features to the capture as upper node.

About the error, can you specify which test should I run?

joansola, 2016-04-08:
The test is test_image. However, you need a video sequence to be able to run it. Maybe you can just go to Jaime and check it out with him.

Thanks for the explanations in this issue. They'll help a lot. I'll try to advance in this today.

joanvallve, 2016-04-08:
It is not necessary to run a test for the moment, I think I found the problem:

In the whole processorTracker environment, we are not using the wolf API (implemented in nodelinked) for adding and removing nodes. We are just modifying directly the std::list<FeatureBase*> of last and incoming captures...

joansola, 2016-04-10:
trackFeatures() now accepts a 3rd argument with a std::map of correspondences. 

Closing.

closed 2016-04-10 by joansola