IRI-MobileRobotics/wolf #49: Wolf's logo

pt07, 2016-04-15:
Is this Wolf's official logo?

[image]

Can I use it is the slide for the thesis defence?

Sorry if I opened this issue, but it is the fastest way to ask you all a question.

joansola, 2016-04-15:
Hi Pier.

The logo is not ours, I took from the internet and it belongs to some other organization or designer. However, I think you can use it in the defence -- I would not use it in your public documents.

Best

pt07, 2016-04-16:
My only public document will be the report, and it's without logos.
If I'll need one to fill the Wolf slide, i'll use this, thank you!

Regards

joanvallve, 2016-04-18:
I quickly made 2 silly logos.. If one of them is good for you we can take it as temporary logo, feel free to propose modifications and for sure to say you don't like them at all! I like the first one

[image]
[image]

joansola, 2016-04-18:
I prefer the first one!
How'd you do them? :-D

andreucm, 2016-04-18:
I like both images. Regarding the text part, I would avoid to mix font types ...

joanvallve, 2016-04-20:
Same font types, more serious: 
[image]

joanvallve, 2016-04-20:
Better?
[image]

joansola, 2016-04-20:
I think I would prefer it without the 'windowed localization frames', just WOLF. Like many other brands, who cares.

Besides, and as Jeremie pointed out, we are not necessarily using 'windows' of keyframes, so the W is no longer justified if not by the fact that we like it like this.

joanvallve, 2016-04-20:
True! I think we are approaching to something simple and cool 
[image]

joansola, 2016-05-03:
I added the logo to the Doxygen generated html page.

joansola, 2016-05-10:
Thanis for this one! Closing as this is no longer an issue!

closed 2016-05-10 by joansola