IRI-MobileRobotics/wolf #50: New branch motion2

joansola, 2016-04-17:
Please @joanvallve and @AtDinesh read this

There is a new branch motion2. This name with the '2' is because Dinesh had an old branch named 'motion' and I did not want to anticipate conflicts. Then:

  * @AtDinesh , are you still having your local branch 'motion'? Do you have interesting code there that needs to be pushed? 
    * If NOT, please:
      * delete your 'motion': ````git checkout master; git branch -D motion````
      * tell me so that I can rename motion2 to 'motion'
      * checkout the new 'motion': ```` git checkout motion````
      * make a branch 'imu' departing from motion: ````git branch imu; git checkout imu````
    * if YES, then
      * let us discuss what code can be merged to motion2, or to master.
      * Then we'll merge motion and motion2, and keep just one of them

  * @joanvallve and @AtDinesh , you should be basing your developments on top of this branch. If you have new branches such as 'imu' or 'laser_motion', please do e.g.:

  ````
    git checkout imu (or laser_moiton)
    git merge motion2
  ````
  so that you benefit from the updates in the motion2 branch. From time to time, when pushes are clean and working, you can merge to motion2 and push:
  ````
    git checkout motion2
    git merge imu
    git push
  ````
  * If you create commits that belong to motion2 (that is, they are not specific of your IMU or LASER things), it is best if you commit them directly to motion2 instead of your current branch:

  ````
    git checkout motion2
    git commit <files to commit> -am 'bla bla'
  ````
I will be merging 'motion2' to and from 'master' when appropriate. I will not touch 'imu' or 'motion_laser' branches.

joanvallve, 2016-04-18:
done

AtDinesh, 2016-04-18:
I have some files that I may need (i'm not done working with them) but I may simply copy them on my computer and add them to the new branch motion.

joansola, 2016-04-18:
Ok

AtDinesh, 2016-04-18:
Just let me know when motion2 will become motion ad I'll do the necessary.

joansola, 2016-04-19:
Actually, I think the branch will stay 'motion2', because renaming involves some unnecessary trouble. Please just  rebase or merge your developments to motion2, and delete your local 'motion'.

AtDinesh, 2016-04-19:
OK
I'm copying files to imu branch created from motion2.

closed 2016-04-20 by joansola