IRI-MobileRobotics/wolf #52: Change WolfProblem to Problem?

joanvallve, 2016-04-18:
In the library, there isn't any other class with "wolf" in the name (apart from wolf_manager, to be removed). After putting the wolf namespace, it is even more redundant... Not urgent.

joansola, 2016-04-18:
The same with WolfConstants (nobody uses but me, it is in wolf.h)

joansola, 2016-04-18:
I do it, together with WOlfConstants. Will push to master some time soon.

I'll post it here when I push

joansola, 2016-04-18:
BTW, also WolfScalar will be renamed.

joansola, 2016-04-18:
Done:
WolfProblem --> Problem (renamed files and #includes accordingly)
WolfScalar --> Scalar
WolfConstants --> Constants

joansola, 2016-04-18:
Pushing to master and closing

joanvallve, 2016-04-26:
getWolfProblem() may be renamed also to getProblem()

joanvallve, 2016-04-26:
In branch laser_motion is already done, when merging it'll be solved

closed 2016-05-10 by joansola