IRI-MobileRobotics/wolf #53: Boost/std::TR1

AtDinesh, 2016-04-18:
I think we wanted not to use Boost at all.. So I am thinking about using std::TR1 (Technical Report 1) for traits.
But TR1 might not be really standard and compiler independent...

Is it ok if I use it or.. any suggestion ?

joansola, 2016-04-18:
Can you find this functionality in c++11? 
Joan Solà - IRI-UPC-CSIC - (sent from cellphone)

AtDinesh, 2016-04-18:
For the traits yes although I don't know if it is exactly the same yet : 
http://en.cppreference.com/w/cpp/header/type_traits

But other features are not (such as concepts implemented in Boost --> I found a way to check concepts but it is also in TR1).

joansola, 2016-04-18:
I have nothing against boost. It is true that we did not want to use it, just to keep simple. But if it comes with great functionality then we can use it.

However, it could be interesting to be able to install just the part of boost we need. Is is possible?

Finally, I do not know what are 'concepts'...

AtDinesh, 2016-04-18:
Maybe it is possible : 
http://www.boost.org/doc/libs/1_60_0/tools/bcp/doc/html/index.html

joansola, 2016-04-18:
What are 'concepts' ?

andreucm, 2016-04-18:
I've never worked with them, but apparently it is something existing in older versions of C++ , that was removed in C++11: https://en.wikipedia.org/wiki/Concepts_%28C%2B%2B%29

AtDinesh, 2016-04-19:
Tags, template, concepts and traits are all part of the generic programming methods (as far as I understood). 
concepts in boost allows one to perform checks on some object to make sure they fit some requirements. If requirements are met then some methods can be defined. It also allows to throw understandable errors.

in my case let's say I have A and B objects.
if A and B are part of additive groups
      then define some methods that can be used (A+B)

This is just an example

artivis, 2016-04-19:
I do not know if you can check the requirements of your objects at compile time but if so you could also have a look at [SFINAE](https://en.wikipedia.org/wiki/Substitution_failure_is_not_an_error).

joansola, 2016-04-19:
Hi. Thank you Dinesh for explanations, and all others for contributions.

I also checked out 'concepts' and I see it is a way to make sure that template parameters meet some requirements at compile time, basically about the functionality one should expect from them. It is true that by now templates leave you with no checks at all. For example, the following template function will fail if we call f() with a double argument:

````
template<class C>
void f(C var){
    var.advance();
}
````

because doubles cannot advance(). With 'concepts', we can have a check on the template class C so that if the compiler sees we are using double it warns us.

So, OK. Like most features in advanced C++ flavors, it is a great thing. And if we need this, then we can incorporate it to wolf. But the questions are: 
  - do we need this?
  - how easy/difficult is to achieve this?
  - how easy it is to explain to others, so that present and future developers of wolf feel comfortable with the code they find?

I am especially concerned with the third question. Because, by now, we are not very skilled in C++ and it represents a lot of effort (for ones more than for others, of course) to incorporate many of these concepts (and we have quite a few already: linux, git, c++, inheritances, virtual functions, abstract classes, templates, object factories, STL containers, ROS, Wolf itself, and of course all the real robotics problems...)

I would therefore ask for wise consideration about incorporating new programming features. We can keep this discussion open and live for a while before we arrive to a conclusion.

Thank you.

AtDinesh, 2016-04-20:
OK we will have to think about this.

Just to make a list of boost features that may become useful in the future : 
- concepts
- serialization

artivis, 2016-04-20:
Regarding serialization, there is also [Cereal](https://github.com/USCiLab/cereal) library.
Being header-based only it could be distributed with Wolf in a 'third-party' folder.

joansola, 2016-05-03:
Closing this discussion

closed 2016-05-03 by joansola