IRI-MobileRobotics/wolf #56: Redesign MotionBuffer by inheriting from list<Motion>

joansola, 2016-04-20:
all is in the title.

joansola, 2016-04-20:
According to this

http://stackoverflow.com/questions/2034916/is-it-okay-to-inherit-implementation-from-stl-containers-rather-than-delegate

it is not a good idea to inherit from list<whatever> to add functionality.

Alternatives:
 * Use typedef:
    ````
    typedef list<Motion> MotionBuffer;
    ````
   and use independent functions to add functionality to MotionBuffer.

 * Put list as member and provide const and non-const reference getter
    ````
    class MotionBuffer {
    private:
        list<Motion> container_;
    public:
        const list<Motion>& get() const  {return container_;}
        list<Motion>& get() {return container_;}
        other_method(...) {... }
    }
    ````
    and use the get() to access all the std::list API. Add new methods as appropriate.

@joanvallve what do you think?

joanvallve, 2016-04-20:
The second alternative is almost what we have now, right? Let's continue like this, then.. removing pushBack() pushFront() etc..

joansola, 2016-04-20:
This is what I did. And pushed to motion2.

closed 2016-04-20 by joansola