IRI-MobileRobotics/wolf #57: Optimize find_if in MotionBuffer

joansola, 2016-04-22:
Hello.

I am trying to optimize the way to find a value of the delta-motion in the buffer of motions. I need to pass a time-stamp, and I want the algorithm to return the last motion just before or equal to this time-stamp.

Now, I have a list, and I use find_if to search each element of the list, starting from the last one and advancing in reverse, until I validate the condition (rev_iterator->motion.ts <= time_stamp).

This works fine, but it is log(n).

I want to use a map<TimeStamp, Motion> to have all elements sorted automatically, ant to allow for a tree search, which is log(n). But I do not want to find the exact key 'time_stamp', but the last one that satisfies the condition.

I found no way to use find_if together with map to obtain log(n). Find_if is always using linear search.

If anyone has an idea of how to achieve this, I'd be very glad to hear!

As a comparison, if the buffer has 100 elements (a reasonable number in our case), a linear search could take up to 100 iterations. A tree search would take at the most log_2(100) < 7 iterations.

AtDinesh, 2016-04-22:
hmmm....

Maybe this could help?
[http://en.cppreference.com/w/cpp/container/map/lower_bound](http://en.cppreference.com/w/cpp/container/map/lower_bound)

joansola, 2016-04-22:
OH!

I did not know this lower_bound and upper_bound things! Thank you!

Yes, ````map<>``` implements the binary tree search, which is log(n), that's why I wanted to use it!