IRI-MobileRobotics/wolf #62: Problem & ProcessorMotion initialization

joanvallve, 2016-04-27:
Mostly, we want to add a initial covariance to the problem. To do this, there is the CaptureFix (we can change its name), adding it to the first frame and processing it, it creates a constraintFix.

But the ProcessorMotion::setOrigin(vectorXs state, ts) creates a key frame that should be the first, also. Maybe it should accept a frame instead of a ts...?

joanvallve, 2016-04-28:
I added a ProcessorMotion::setOrigin(FrameBase*, ts) to be used in case the first keyframe is already created before the processor motion initialization.

However, it is still undefined the wolf problem initialization procedure, so I leave this issue open...

joanvallve, 2016-05-04:
There are 2 things that must be done in order to initialize the problem and the `processMotion`:

1. `Problem::setProcessorMotion(ProcessorMotion*)`
2. `ProcessorMotion::setOrigin(FrameBase*, TimeStamp)`

This may be related with issue #58

joansola, 2016-05-04:
I think we should only provide one function call for initialization, in Problem. the rest should be automatic. Can you come up with an API proposition that makes sense from the user point of view? Please consider #58

joansola, 2016-05-04:
IMO, we could do:
 - ````Problem.setProcessorMotion("Processor name")````.
 - ````Problem.setOrigin(FrameBase frame_0)````
 - ````Problem.setOrigin(const TimeStamp& ts, const VectorXs& x_0)````

joanvallve, 2016-05-10:
I think that the first frame is more an input parameter in the problem constructor (with the option to provide an initial covariance too) and the processorMotion should take it as origin when it is added (or setted or installed... XD)

joanvallve, 2016-05-11:
I'm moving wolf tests to the factory API. These two things should be discussed:

**A. Initial pose**
There should be a `problem` constructor accepting an `_init_pose` and `_init_covariance`, executing the following:

1. `origin_frame_ptr = createFrame(KEY_FRAME, _init_pose, ts)`
2. `SensorFix2D* fix_sensor_ptr = createSensor("FIX_2D_POSE", "initial pose")`
3. `CaptureFix* init_capture = new CaptureFix(ts, fix_sensor_ptr, _init_pose, _init_covariance)`
4. `origin_frame_ptr->addCapture(initial_covariance)`
5. `init_capture->process()`

**B. ProcessorMotion initialization**
Initializing the processor motion means 2 things:
- `ProcessorMotion::setOrigin(_origin_frame_ptr)`
- `Problem::setProcessorMotion(_processor_motion_ptr)`

I think both should be done in the _installation process_...

Other alternative is doing **A** and **B.1** in something like:
`Problem::initialize(_init_pose,_init_covariance)`
But it should be called after the installation of the motion sensor.

joansola, 2016-05-11:
Good point.

Can we guarantee integrity no matter the order in which we initialize things? I think this should be doable:
- If problem.setOrigin() is after problem.installProcessor(), we need to initialize the processor just after installing it, in the installer.
- If problem.installProcessor is after problem.setOrigin(), we need to propagate the initialization down to the processor.

joanvallve, 2016-05-11:
So to be sure, do you mean that you would discard the alternative of setting the origin in the problem constructor?

Then, in the `installProcessor` there should be a `Problem::setProcessorMotion(_processor_motion_ptr)` call. I see two problems:
- Is there a way to use polymorphism to force that processor motion installers should call `Problem::setProcessorMotion(this)`?
- Dealing with more than one processor motion: The processor motion installer should accept somehow specifying if the `Problem::setProcessorMotion(this)` should be called.

joansola, 2016-05-11:
No no, I just say that whatever the order of initializations, everything should end up OK. Then I put an example, but I did not think a lot about it.

joansola, 2016-05-11:
The one processor motion needs to be known by problem, and it's one of the parameters of the configurations. I think specifying the origin means:
- set mean and cov
- set which is the processor motion

and that this should be done only once, for example in problem:
- problem.setOrigin(mean, cov)
- problem.setProcessorMotion("Processor Name")

but, again I did not think about it enough...

joansola, 2016-05-11:
Otherwise, we can enforce an order. I see this as the logical order:

- problem.installSensor()
- problem.InstallProcessor()
- problem.setProcessorMotion("Name")
- problem.setOrigin(mean,cov)

joanvallve, 2016-05-11:
My point is that if the origin is setted in the problem constructor, then we force the order (we can't install anything in a problem that is not created yet).

However, I'm not sure if always in the constructor call time, the final user will always know the origin..

Let's do it separately. I'll figure out if the order is really an issue.

joanvallve, 2016-05-11:
I think we can avoid your 3rd point if when installing the processor motion, setProcessorMotion is automatically done. We can specify that, for instance, the first processor motion installed is the main one.