IRI-MobileRobotics/wolf #63: Eigen, SuiteSparse, etc

joansola, 2016-04-27:
Hi, 

On my new Mac, I am having trouble with Eigen and the sparse routines, I think coming from SuiteSparse (not sure).

I particular, I cannot compile test_iQR and similar.

My error is the following:

````
In file included from /Users/jsola/dev/wolf/src/examples/solver/test_iQR_wolf.cpp:21:
In file included from /usr/local/include/eigen3/Eigen/OrderingMethods:4:
In file included from /usr/local/include/eigen3/Eigen/SparseCore:48:
/usr/local/include/eigen3/Eigen/src/SparseCore/SparsePermutation.h:64:127: error: 'nonZeros' is a private member of 'Eigen::BlockImpl<const Eigen::Block<Eigen::SparseMatrix<int, 0, int>, -1, -1, true>, -1, 1, true, Eigen::Sparse>'
          sizes[((Side==OnTheLeft) ^ Transposed) ? jp : j] = m_matrix.innerVector(((Side==OnTheRight) ^ Transposed) ? jp : j).nonZeros();
                                                                                                                              ^
/usr/local/include/eigen3/Eigen/src/Core/ReturnByValue.h:61:42: note: in instantiation of function template specialization 'Eigen::internal::permut_sparsematrix_product_retval<Eigen::PermutationBase<Eigen::PermutationMatrix<-1, -1, int> >, Eigen::Block<Eigen::SparseMatrix<int, 0, int>, -1, -1, true>, 2, true>::evalTo<Eigen::SparseMatrix<int, 0, int> >' requested here
    { static_cast<const Derived*>(this)->evalTo(dst); }
                                         ^
/usr/local/include/eigen3/Eigen/src/SparseCore/SparseMatrix.h:679:13: note: in instantiation of function template specialization 'Eigen::ReturnByValue<Eigen::internal::permut_sparsematrix_product_retval<Eigen::PermutationBase<Eigen::PermutationMatrix<-1, -1, int> >, Eigen::Block<Eigen::SparseMatrix<int, 0, int>, -1, -1, true>, 2, true> >::evalTo<Eigen::SparseMatrix<int, 0, int> >' requested here
      other.evalTo(*this);
            ^
/usr/local/include/eigen3/Eigen/src/SparseCore/SparseView.h:40:7: note: in instantiation of function template specialization 'Eigen::SparseMatrix<int, 0, int>::SparseMatrix<Eigen::internal::permut_sparsematrix_product_retval<Eigen::PermutationBase<Eigen::PermutationMatrix<-1, -1, int> >, Eigen::Block<Eigen::SparseMatrix<int, 0, int>, -1, -1, true>, 2, true> >' requested here
    : m_matrix(mat), m_reference(reference), m_epsilon(epsilon) {}
      ^
/usr/local/include/eigen3/Eigen/src/SparseCore/SparseView.h:94:10: note: in instantiation of member function 'Eigen::SparseView<Eigen::ReturnByValue<Eigen::internal::permut_sparsematrix_product_retval<Eigen::PermutationBase<Eigen::PermutationMatrix<-1, -1, int> >, Eigen::Block<Eigen::SparseMatrix<int, 0, int>, -1, -1, true>, 2, true> > >::SparseView' requested here
  return SparseView<Derived>(derived(), m_reference, m_epsilon);
         ^
/Users/jsola/dev/wolf/src/examples/solver/test_iQR_wolf.cpp:294:133: note: in instantiation of member function 'Eigen::MatrixBase<Eigen::ReturnByValue<Eigen::internal::permut_sparsematrix_product_retval<Eigen::PermutationBase<Eigen::PermutationMatrix<-1, -1, int> >, Eigen::Block<Eigen::SparseMatrix<int, 0, int>, -1, -1, true>, 2, true> > >::sparseView' requested here
                    A_nodes_.rightCols(ordered_nodes) = (A_nodes_.rightCols(ordered_nodes) * partial_permutation_nodes.transpose()).sparseView();
                                                                                                                                    ^
/usr/local/include/eigen3/Eigen/src/SparseCore/SparseBlock.h:82:11: note: declared private here
    Index nonZeros() const;
          ^
````

I found this link on te internet:

https://eigen.tuxfamily.org/dox-devel/group__SPQRSupport__Module.html

maybe we need to think which is the best way to have the sparse matrix functionality.

joanvallve, 2016-04-27:
Which version of eigen are you using?

This error comes from eigen. It appears when a permutation matrix is multiplied by a sparse matrix... I don't understand why are you having this error and not in linux. 

But it is not directly related neither with the SPQRSupport module you link nor SuiteSparse.

joansola, 2016-04-27:
Ah, OK. Good that I asked, I see you know more than me. I could not find sensible information on the internet.

I have Eigen 3.2.8

joanvallve, 2016-04-27:
Can compile the other tests in examples/solver?

joansola, 2016-04-27:
Do you have any idea of how can I get rid of it?

joanvallve, 2016-04-28:
I've made explicit that SparseMatrices are Col Major in that file (test_iQR_wolf.cpp). I think it is somehow related with that.. 

If it doesn't work, you can bring your Mac or maybe I can upgrade my Eigen's version and see if it still happens

joansola, 2016-04-28:
OK we'll see it. I try to fix it myself first.