IRI-MobileRobotics/wolf #69: FEAT_CORNER should be renamed to FEAT_CORNER_2D ?

andreucm, 2016-05-05:
It is a minor comment, but just to be consistent, should we add "_2D" when in 2D case ?

Because I added FEAT_LINE_2D for 2D line segment features.

At wolf.h , 

typedef enum
{
    FEAT_CORNER,
    FEAT_FIX,
    FEAT_GPS_FIX,
    FEAT_GPS_PR,
    FEAT_ODOM_2D,
    FEAT_MOTION,
    FEAT_POINT_IMAGE, 
    FEAT_LINE_2D
}FeatureType;

The issue is just about following a criteria .... The same would apply for LandmarkType

joansola, 2016-05-05:
There are a few oddities in these enums, not only FEAT_XXX but also others. They should follow a strict rule, otherwise we have conflicts:
SEN_LIDAR should be SENSOR_LASER_2D because the class is SensorLaser2D
SEN_GPS_RAW should be SEN_GPS because the class is SensorGPS
etc...
The enum should match exactly the class name so that there is no ambiguity. 

then, coming back to the question, see if also the class need to have this "2D" added.

joansola, 2016-05-06:
@andreucm you can go ahead and rename the enums! Just do it for all the library!

andreucm, 2016-05-11:
Fixed FEAT_* to FEATURE_*
Pushed to branch motion_laser

Still to decide about "_2D" or "_3D" postfix. 
Still to fix other EnumTypes

joansola, 2016-05-13:
" Still to decide about "_2D" or "_3D" postfix. " you mean #48 ? It's uppercase D

andreucm, 2016-05-13:
Yes , it's always uppercase... but I was meaning when to add it or not, ...

joansola, 2016-05-13:
I guess you need to add it everywhere it is needed. Rules (proposal) :
- If what you coded is 2D, then write it. If it can be either 2D or 3D, then leave it blank. 
- Of course, this only applies to cases where ambiguity may exist. 
  - Laser and Odometry I think both need the D decorator
  - Cameras and IMUs do not
  - GPS should be always 3D in my opinion. 
    - If there is one 2D implementation, then mark it. But the 3D i would not mark it