IRI-MobileRobotics/wolf #70: General issue on constructors, factories and optional libraries.

joansola, 2016-05-05:
Hi.

I have found a bug in the architecture that needs to be fixed. It is not easy to explain, I'll try. Please read carefully.

We have optional libraries. For example, ````libraw_gps_utils````. If the library is found, Wolf takes the related files:
````
IF (raw_gps_utils_FOUND)
    SET(HDRS ${HDRS}
        capture_gps.h
        constraint_gps_2D.h
        constraint_gps_pseudorange_3D.h
        constraint_gps_pseudorange_2D.h
        feature_gps_pseudorange.h
        processor_gps.h
        sensor_gps.h
        )
    SET(SRCS ${SRCS}
        capture_gps.cpp
        feature_gps_pseudorange.cpp
        processor_gps.cpp
        sensor_gps.cpp
        )
ENDIF(raw_gps_utils_FOUND)
````
and adds them to the ````lobwolf.so```` library. Fine until here.

We also have wrappers that need to create objects out of wolf that interface wolf with other tools. In particular, we have, in ````create_autodiff_cost_function.h````:
````
#include "../constraint_gps_2D.h"
#include "../constraint_gps_pseudorange_3D.h"
#include "../constraint_gps_pseudorange_2D.h"
    switch (_ctr_ptr->getType())
    {
        case CTR_GPS_PR_3D:
            if (_use_wolf_autodiff)
                return createAutoDiffCostFunctionWrapper<ConstraintGPSPseudorange3D>(_ctr_ptr);
            else
                return createAutoDiffCostFunctionCeres<ConstraintGPSPseudorange3D>(_ctr_ptr);

        case CTR_GPS_PR_2D:
            if (_use_wolf_autodiff)
                return createAutoDiffCostFunctionWrapper<ConstraintGPSPseudorange2D>(_ctr_ptr);
            else
                return createAutoDiffCostFunctionCeres<ConstraintGPSPseudorange2D>(_ctr_ptr);
````
which is not going to compile properly, should the GPS library be not installed.

Now, this is just an example, which is real. In our cases, and you can check it out, the ````CMakeLists.txt```` adds this line in the plain ````$SRC```` definition:

    ````sensor_gps.cpp````

which is out of the guards of the ````raw_gps_utils_FOUND```` stuff. As you can see, this file is here so that, even if the library is not there, we compile it, and then the rest of the code, magically (dont ask me why!) also copiles.

But, this cannot be allowed. If the library is not there, then none of these files should ever be considered.

This brings us to the code in the ````create_autodiff_cost_function.h```` file: it is a factory of cost_functions, and it should construct only the kinds of elements that are installed, not others.

Therefore, I see that we will have to remove all of the factories made with ````switch / case```` code, which are static, and replace them by proper factories where the optional elements can be dynamically registered. The design of these factories would be the same as the one used for Sensors and Processors.

So, summarizing, we should make a point and discover in how many places we used this ````switch/case```` method to create elements dynamically, and, if these elements can contain cases in optional libraries, then build the correct factory. 

Sorry but I think this is a serious issue. If you have all the libraries properly installed, then you do not feel the need for it. But the issue will raise up if you uninstall them.

I wish I was wrong about this. To be sure, can you please make a test ? It is the following:

 - git checkout master
 - comment out sensor_gps.h in src/CMakeLists.txt
 - go to the raw_gps_utils library
 - make uninstall
 - go back to wolf
 - make

See if it works. Then, again with 'factory'
 - git checkout factory
 - comment out sensor_gps.h in src/CMakeLists.txt
 - make

Can you report your results here? Thank you.

joansola, 2016-05-05:
Uhm... I fixed this, but I am not convinced of my solution. We'll discusse it live.

joansola, 2016-05-05:
Maybe I am having issues because my system does not find stuff installed in 
````
/usr/local/include/iri-algorithms...
/usr/local/lib/iri-algorithms/...
````

@andreucm : How do you tell the system that it needs to go to iri-algorithms ? I fixed one include, but what about the library itself? Why is so that it works automatically in ubuntu and not in Mac?

joanvallve, 2016-05-06:
I think theses folders are setted in the FindXXX.cmake, there are the folders where to look for the library...

I've checked what you asked and I have the same undefined reference wolf::SensorGPS::create() in both branches master and factory.

With respect to the dynamic factories.. I'm not sure which is your solution, the cost function creation needs to be static (templated to the constraint type), however it doesn't need to be in the same file... is it enough?

joansola, 2016-05-06:
@joanvallve also in master? Oh, no way! 
I pushed a cleaner version in both factory and master. Can you re-try?

joansola, 2016-05-06:
As for the dynamic factories, we need to discuss calmly about it. Identify what do we need, and then do it. Now I am quite good at designing factories and giving them a friendly API! :-D

artivis, 2016-05-06:
I'm experiencing a similar issue. Since I have OpenCV 3 on my laptop, almost everything that uses OpenCV crashes at compilation on my machine.
The OpenCV guards in the `CMakeLists.txt` such as `OpenCV_FOUND` should also check for the version found.

joansola, 2016-05-06:
I guess those issues are not new, and that they do not interfere with your present work. Am I wrong? Or are we facing an urgent situation?

I other words:
Can you compile master?

joansola, 2016-05-06:
Personally, I hate CMAKE (just because I do not understsand it). But I agreee that it might need a revision, with the addition of convenient guards and checks. This includes:

- OpenCV 3
- Mac OSX
- SuiteSparse, SparseBlocks, as in #63 
- Optional libraries (laser, gps) as in this issue #70

artivis, 2016-05-06:
As a matter of fact I can't because of opencv changes. I simply comment opencv find_package for now.
It is simple to call find_package for a given version of opencv.

artivis, 2016-05-09:
Actually really simple :

In CMakeLists.txt replace
`FIND_PACKAGE(OpenCV QUIET)`
with
`FIND_PACKAGE(OpenCV 2 QUIET)` .

@joansola I let you make this change directly to master, that doesn't worth a merge request :p

joansola, 2016-05-09:
done and pushed to master

closed 2016-05-09 by joansola