IRI-MobileRobotics/wolf #73: API for contraints: weyword 'const'

joansola, 2016-05-07:
See if we use const in the Constructors of constraints. Now the accepted pointers are not const. This is undesirable, but I ignore if this is a requirement of some other part of the code, e.g. the solver or something.

joansola, 2016-05-07:
See e.g. here:
````
		ConstraintCorner2D(FeatureBase* _ftr_ptr, LandmarkCorner2D* _lmk_ptr, ConstraintStatus _status = CTR_ACTIVE) :
			ConstraintSparse<3,2,1,2,1>(_ftr_ptr, CTR_CORNER_2D, _lmk_ptr, _status, _ftr_ptr->getFramePtr()->getPPtr(),_ftr_ptr->getFramePtr()->getOPtr(), _lmk_ptr->getPPtr(), _lmk_ptr->getOPtr())
		{
		    //
		}
````

joanvallve, 2016-05-09:
What are you proposing, const pointer or pointer to const object? For the first case I don't see a problem, for the second... a lot.

joansola, 2016-05-09:
I don't know :-D I guess const pointer... how would it be?
- ````const FrameBase * p```` <-- NOT this one: "pointer to constant FrameBase"
- ````FrameBase const * p```` <-- NOT this one: "pointer to constant FrameBase"
- ````FrameBase * const p````  <-- this one: "constant pointer to FrameBase"

joanvallve, 2016-05-10:
Yes, the first and the second are equivalent, right?
However... which sense has passing const pointers if they are not passed by reference?

joansola, 2016-05-10:
I don't know... I just thought there should be a ````const```` in the API... like when passing so many other params to configure and create things...

joansola, 2016-05-10:
It is not very important. I just noticed it and raised the issue. But we can live with this, especially considering that pointers should be const only if passed by reference, which is not very useful.

So, I think I'll close this issue after you give an OK.

joanvallve, 2016-05-10:
OK!

closed 2016-05-10 by joanvallve