IRI-MobileRobotics/wolf #75: Improve API for ConstraintContainer constructor

joansola, 2016-05-08:
It has a parameter ````_corner```` which impacts the operation of the ````ConstraintFactory```` negatively. It is the only Constraint to require such extra parameters. All other Constraints accept:

````ConstraintXxx (FeatureBase*, NodeBase*, ConstraintType)````

where ````NodeBase*```` is here to be able to accept the three alternative pointers, ````FrameBase*````, ````FeatureBase*````, and ````LandmarkBase*````. In case of absolute constraints, this pointer can be passsed as ````nullptr````.

- Can we avoid this parameter ````_corner````?
- Can this parameter be in the Feature that will own the Constraint, so that passing a pointer to the Feature (as it is done) allows the constructor to retrieve the parameter? A solution like this exists already in ````ConstraintGPSPseudorange2D```` and ````ConstraintGPSPseudorange3D````.

joanvallve, 2016-05-09:
No problem with this new API. This parameter can be in the Feature. However, this is deprecated since the new ProcessorTracker implementation and waiting for SuperLandmarks implementation to be implemented

joansola, 2016-05-09:
If deprecatred, can it be removed from wolf?

joanvallve, 2016-05-09:
I propose to remove it in the branch motion_laser, where we are applying the new laserscanutils API and the new ProcessorTracker for laser

joansola, 2016-05-09:
I changed the API so as to accept only feature pointer and correspondant pointer, with no arguments.

We'll remove obsolete objects at due time. You can do it in your branch if you wish.

joansola, 2016-05-10:
closing this one.

closed 2016-05-10 by joansola