IRI-MobileRobotics/wolf #76: Plugins in WOLF: segmenting the library into separate and independent plugins

joansola, 2016-05-09:
A conversation with Nicolas Mansard at LAAS, about the organization of the wolf library:
__NM__: Nicolas Mansard
__JS__: Joan Sola

__NM__: Be careful with one last aspect: normally, factories are to be used as plugins, i.e. each new class (or group of classes) implementing the API is in a dedicated .so file, which is dynamically loaded *after* the main library (containing the factory) is loaded.

__JS__: OK I see. But what do you mean with 'factories are to be used as plugins' and ' *after* the main library (containing the factory) is loaded'? This is a contradiction no? You mean that 'each new class (or group of classes) implementing the API' is a plugin, therefore is in a dedicated .so file, but hte factory is in the main library. This is how I understand it. OK?

__NM__: Yes. I mean: the main library (libwolf.so) should only contain the factory (i.e. API, abstract classes, factory map, plugin loader), and each class (e.g class SensorCamera) should be in a dedicated plugin (sensorcamera.so) to be loaded later.
After compilation you would have three objects:
libwolf.so (contain the factory)
main.exe (contain the function main, is linked with ldd with libwolf.so and call the plugin loader methods -- dlopen).
sensorcamera.so (not linked with main.exe in any direction, but linked with libwolf.so, to be loaded explicitely inside the main).

joansola, 2016-05-09:
I think this is becoming important: WOLF is now huge, and contains work that can be treated independently as different plugins:
- Odometry 
- IMU 
- Camera, image, etc, 
- Laser 2D
- Laser 3D

artivis, 2016-05-09:
If splitting wolf into plugins per sensor type make sens, what about processors ?

joansola, 2016-05-09:
The idea is that there may exist bundles of objects related to a particular sensor:
- The sensor
- the Capture
- the processor
- a particular Feature and / or Landmark type
- the consraints

It is known that, sometimes, or even often-times, this is not as simple, as a Landmark can be used in different processors, for example. Then, maybe, we can forget about the bundles, and make a .so plugin for each object that does not belong to the base API of wolf -- i.e., every specific object is shipped separate from the WOLF library in its own .so file.

joansola, 2016-05-09:
I have no idea of how to load the plugins,
- at link time
- at run time

but by now I am content to know that:
- it is doable
- it is not difficult to do
- it is not difficult to document

nmansard, 2016-05-10:
I would be in favor of bundles. Note that you can have a plugin that depends on another plugin, just remember to load by order of dependency.

Plugins and software projects are different granularity. You may have several plugins inside the main wolf project. Splitting the code in several plugins should not have a deep impact. Splitting the project in several small projects is generally seen as a good point, but is also heavier to handle and manage, especially for academic software. 

Loading a .so file is very easy: just call the function dlopen with the path to the .so file. See the howto below:
http://tldp.org/HOWTO/Program-Library-HOWTO/dl-libraries.html
For factory plugins, loading the corresponding .so file will execute the registration code in the .cpp file, and automatically add the classes inside the factory. A bundle of classes will typically corresponds to several .cpp, compiled in several .so, and linked in a single .so file which has to be dynamically loaded.

So: it is doable, not difficult to do (with minor subtle aspects with the loading flags) and straightforward to document.