IRI-MobileRobotics/wolf #78: Loss function in wolf

joanvallve, 2016-05-11:
Ceres accepts a loss function independently for each constraint. Are we going to use it? It could be useful for outlier rejection.

Until now, in the moment of updating the ceres, a boolean adds a specific loss functions with hardcoded parameters for all new constraints.

If we want to keep this functionality, something should be added to the constraintBase node. For now, I will move this boolean to the constraintBase level. Then, we can remove it or develop it.

joansola, 2016-05-11:
What I see is that we will detach from Ceres at one point. We should be careful when adopting Ceres features, because it'll be very difficult to pass without it afterwards!

joanvallve, 2016-05-12:
A loss function can/may be available in other solvers. If there isn't this option, we can just ignore this flag in other solvers wrappers..