IRI-MobileRobotics/wolf #80: Factories

joansola, 2016-05-22:
Hello. Factories for the following objects are ready and working in branch 'master':
 - ````SensorXxx````
 - ````ProcessorXxx````
 - ````IntrinsicsXxx````
 - ````ProcessorParamsXxx````

It is now interesting that you take your objects and make them factory-enabled. Basically, using these factories serves to create these objects easily from information provided either from text, of from configuration files. This is useful in the ````main()```` functions of our applications, or in the ROS nodes.

It is therefore convenient that you upgrade your main() and ROS nodes to make use of this functionality.

Please look at ````test_yaml.cpp````, ````test_wolf_factories.cpp````, and ````test_image.cpp```` for examples of use. In particular, look at the way to add sensors and processors to ````Problem````:
 - ````installSensor()````
 - ````installProcessor()````

Please read the documentation in ````sensor_factory.h```` for a thorough explanation of the use of the factories. In particular, you need to ensure that:
 - You create methods ````SensorXxx::create(...)```` and ````ProcessorXxx::create(...)```` to create sensors and processors from the factory. Look at ````SensorCamera```` and ````ProcessorImage```` for complete examples.
 - Register these methods to the factory. Look at ````sensor_camera.cpp```` for an example of a registration instruction.
 - You create methods ````createIntrinsicsXxx(...)```` and ````createProcessorParamsXxxx(...)```` to create parameters for your objects from YAML or XML files. Look at ````yaml/sensor_camera_yaml.cpp```` for en example. Look also at ````yaml/processor_image_yaml.cpp````.
 - You register these methods into the factories. See ````.cpp```` files in the directory ````yaml/````

joansola, 2016-05-22:
@JaimeTarraso I completed most of the actions for your classes to use the Factory. You should now use it in the ROS node. Please look at test_image.cpp to see what you have to do.

@joanvallve You'll probably have to merge master to you branch before.

@AtDinesh you probably do not need to bother about this by now. If you do, you can ask Nicolas in case of any difficulties -- he is aware of the design of the Factories.

joanvallve, 2016-05-23:
Have you removed the processor installer from the sensor name and the instrinsics ptr?

joansola, 2016-05-23:
Yes, because it lies halfway between the all-code API and the all-text API and I thought it was misleading. Do you need it, or can you work with the current API?

joansola, 2016-05-23:
By now, the way to bind captures to sensors is by returning a ponter to the sensor in installSensor(). However, you can obtain this pointer from a sensor name using problem.getSensorPtr("Sensor Name")

joanvallve, 2016-05-23:
I like the all-text API, however, in ROS it is not aplicable.. right? BTW: merged and working in tests and node ;)

joansola, 2016-05-23:
The IntrinsicsFactory and the ProcessorParamsFactory can be loaded with any creator method that agrees with this:

````IntrinsicsBase * create(std::string type, std::string filename);````

Therefore, it can be used with ROS' rosparams probably.

At this time, creators from YAML files (wolf style) are automatically registered to the factory. But this will be removed in the future, and registration will be done according to the application needs (WOLF or ROS or other). In fact, each set of params can be created with a different method: just register each creator as desired.

joanvallve, 2016-05-23:
I don't know which way will be the best..

I found an error (i think), in `Problem::installProcessor`, there is the option of not passing the parameters (it is `nullptr` by default), but then it is used in `ProcesorFactory::create`. So if you call `installProcessor` without the parameters, it doesn't compile.

joansola, 2016-05-23:
The default is for constructors that do not need parameters. If you need parameters, then you cannot give a nullptr.

joanvallve, 2016-05-23:
Sorry..
Yes I know, in ProcessorOdom2D, I used this option. But this morning the compiler complained about line 41 in processor_factory.cpp where this is used. But not now, it compiles. Weird..