IRI-MobileRobotics/wolf #81: Set project variable, use it in code

joansola, 2016-05-23:
A question. I would like to have CMAKE to set a project path to the wolf project, such as e.g.

    SET(ENV{WOLF_DIR} ${CMAKE_CURRENT_SOURCE_DIR})

this would set WOLF_DIR to /home/jsola/dev/wolf/ in my case, and make it an environment variable.

Then, using std::getenv(), I would ike to write in my main()

    char* wolf_dir = std:getenv("WOLF_DIR")

so that I can now refer to the YAML configuration files relative to the project root path.

The problem is that this does not work. The SET(ENV{...} bla bla) does not set the variable as an environment variable, and therefore I cannot use it in getenv().

Any ideas? @artivis , how do you solve this at PAL?

joansola, 2016-05-24:
Preliminary solve:

__Problem:__ we need a platform-independent way to specify where are our configuration files
__Cause:__ YAML::LoadFile needs an absolute path to a filename
__Solution:__
  1. To run from Terminal, you need to create an environment variable WOLF_ROOT pointing to where the wolf project is
      - Edit file ~/.bashrc, or ~/.bash_profile, and add these lines:
    
    ````
# WOLF
export WOLF_ROOT="/abs/path/to/wolf"
    ````

  2. If you run your application from eclipse, do:
     - Menu Run > Run configurations...
     - Add or edit a run configuration for the executable you want to run
     - Click on tab 'Environment'
     - Add a variable named WOLF_ROOT, with value /abs/path/to/wolf

  3. If you run from QtCreator
     - I do not know. Possibly it works once 1. is done, otherwise get inspiration from 2.
     - If you succeed, please post here your solution!

AtDinesh, 2016-05-24:
For QT, a solution may also be to use the "Build Environment" part in the Build Settings.
Here should be able to add a variable but I do not know if it will work to run the code.

joansola, 2016-05-24:
Thanks Dinesh

I found it. I edit the instructions above.

joansola, 2016-05-25:
Closing. See documentation on main README.md file in master.

closed 2016-05-25 by joansola