IRI-MobileRobotics/wolf #82: Trajectory: Mantain keyframe order by timestamp

joanvallve, 2016-05-24:
For different reasons (fixing [this bug](https://github.com/IRI-MobileRobotics/wolf/commit/762afd5c5daf0f2fa5c147bde71dd5a1dcadc187#commitcomment-17590332), for instance) it is very useful having, at least, the keyframes ordered in the trajectory `down_node_list_`. 

I see 2 solutions:

1. Mantaining the order in Frame creation and managing.
2. Sort the list using `sort()` (cost NlogN) after a change.

Regular frames timestamps are constantly changing in `ProcessorTracker::process()`. If we want to keep ordered all frames, in both solutions, it would be necessary to reorder a frame each time its timestamp is changed. It can be too much.. We can try to keep only keyframes ordered. 

In **solution 1**, it would imply modifications in:
- `Trajectory::addFrame`: (Only if the added frame is key)  it should be inserted in the right position.
- `FrameBase::setKey`: the frame pointer should be moved to the right position. 
Both modifications are related with the` NodeLinked::down_node_list_` management. So new nodeLinked functions are needed (dealing with down_node_list in FrameBase is not a good option.. right?).

In **solution 2**, it's only calling `sort()` after the two previous cases and adding a comparision function between frames. It would order also regular frames but until next `sort()` call, their order wouldn't be granted.

If we only keep the keyframes ordered, we would call `sort()` only when a new keyframe is created, I think we can afford it, right?

For now, I'm going to implement the second option. It's simpler. If we decide to implement the first one, it will also be easy to undo. Do you see more solutions?

joansola, 2016-05-24:
I think this must be ensured only on Frame addition. What about finding the right position to add the frame, then add it with `insert`? It probably lies very close to the last position, no? This way, we never need to call `sort`.

joansola, 2016-05-24:
Sorry, my previous answer is bull***

If we only want to order keyframes, then we may want a list of frames and a list of keyframes...

In any case, when calling setKey(), we can just find the position and change it, not call sort, because only the frame under consideration will be out-of-sort. Moreover, its new, sorted position will be quite close to the current one. Possibly, calling `sort` will be too much.

We can keep a pointer or iterator to the last frame set as keyframe, and start a position search from there. the search would be linear. The direction of search would be determined by the sign of TimeStamp1 - TimeStamp2...

joansola, 2016-05-24:
Another thing: regarding operations in NodeLinked::down-node-list. T would say that these methods should be in Trajectory, not in NodeLinked. Only in Trajectory the children are Frames, and therefore it makes sense to ask for their TimeStamp. General chilldren of NodeLinked do not have TimeStamp. The API of NodeLinked would be not good if these new methods were added to NodeLinked.

joansola, 2016-05-24:
And clearly I would **not** put the methods in Problem. Just in case.

joanvallve, 2016-05-24:
I do not want to implement these methods in `NodeLinked`, but I don't want to implement down_node_list manipulations (insert, or reordering) in `TrajectoryBase` or `FrameBase`, either.

So something like 'insertDownNode()` and `moveDownNode()`should be implemented in `NodeLinked` and then called from `TrajectoryBase` ordering implementation.

Anyway, as I said, the sort must be done not only in `setKey()` but also in `addFrame()` in case the frame added is key.

I know that `sort()` is too much. But since it is done only after a new key frame is created, I thought it wasn't a big deal. Ok, an extra TODO in wolf! ;)

joansola, 2016-05-24:
OK for having insertNode and moveNode in NodeLinked.

As for the sort(), you need to implement it anyway, because you will sort over only a member of FrameBase, no? I mean, you cannot call list::sort() alone. Then, something like (this is dummy code but still):

````
if (Frame.TimeStamp < lastKeyFrame.TimeStamp)
    auto r_it = find_if (
        Trajectory.getFrameList(),    // list to search for
        rev_iterator(lastKeyFrame),   // first element, in reverse order
        getFrameList().rend(),        // last element
        [&](FrameBase* f){            // the search condition (see lambda function in C++11)
            return f->getTimeStamp() < Frame.getTimeStamo(); 
        } 
    );
    getFrameList()->insert(Frame, r_it)
else
    getFrameList()->addFrame(Frame)
````

joanvallve, 2016-05-24:
I already implemented the sort with a lambda function, but now I'm implementing the cheaper alternative.

joansola, 2016-05-24:
Oh! You had already done it!! you are so quick!

I have edited the algo a little... we need to store an iterator to the last keyframe, and keep it updated.

joanvallve, 2016-05-24:
In TrajectoryBase there is a pointer to the last key frame, wasn't it enough?

joansola, 2016-05-24:
I think you do better with the iterator, not a pointer to the frame...

joansola, 2016-05-24:
The best then is to keep a **reverse** iterator to the last keyframe, and use it and update it directly...,