IRI-MobileRobotics/wolf #83: Dependence on ProcessorMotion

joansola, 2016-05-24:
This issue follows the discussion in commit 762afd5

After some work on the processor trackers side, we created a dependency on having to have a processorMotion in the system, even if we are just playing with a feature tracker. 

This issue has been discussed in commit 762afd5, and this discussion will follow here. Please read the coments on commit 762afd5 for reference.

joanvallve, 2016-05-24:
As I said, I imagine two solutions:
1. Setting a default processor motion (constant velocity, for example)
2. In Problem::getStateAtTimeStamp(ts), return the last keyframe state or just zero state.

The second one is better. Is more generic and it accepts the future implementation of the first one.

joansola, 2016-05-24:
Oh I see. Sorry I am not paying much attention. Your solution 2 seems pretty much the same of my suggestion...