IRI-MobileRobotics/wolf #88: Change dynamic Eigen vectors & matrices with known size to fixed size

joanvallve, 2016-07-11:
To improve efficiency, all matrices and vectors with known size in compile time should be changed to fixed size (only for small matrices ~up to 16)

joansola, 2016-07-12:
Please run the test test_matrix_prod to see computing speeds. Use 'release' compilation.

joansola, 2016-07-12:
Results on my Mac:

````
vermicelle:bin jsola$ ./test_matrix_prod 
Timings for dynamic matrix product. R: row major matrix. C: column major matrix. 3x3 matrices.
      -0.178705	Time Ro = R * R: 562.07ns
       0.735986	Time Ro = R * C: 649.27ns
      -0.390624	Time Ro = C * R: 611.07ns
       0.413712	Time Ro = C * C: 588.5ns
      -0.178705	Time Co = R * R: 569.41ns
       0.735986	Time Co = R * C: 610.38ns
      -0.390624	Time Co = C * R: 580.58ns
       0.413712	Time Co = C * C: 582.86ns <-- this is the Eigen default!
Timings for dynamic matrix product. R: row major matrix. C: column major matrix. 4x4 matrices.
      -0.994905	Time Ro = R * R: 536.67ns
      -0.179013	Time Ro = R * C: 556.12ns
       -1.89683	Time Ro = C * R: 525.39ns
       0.920567	Time Ro = C * C: 554.49ns
      -0.994905	Time Co = R * R: 522.82ns
      -0.179013	Time Co = R * C: 541.42ns
       -1.89683	Time Co = C * R: 510.87ns
       0.920567	Time Co = C * C: 533.53ns <-- this is the Eigen default!
Timings for dynamic matrix product. R: row major matrix. C: column major matrix. 5x5 matrices.
      -0.168334	Time Ro = R * R: 597.68ns
      -0.248837	Time Ro = R * C: 629.97ns
       -3.45701	Time Ro = C * R: 593.83ns
       0.706489	Time Ro = C * C: 647.02ns
      -0.168334	Time Co = R * R: 587.21ns
      -0.248837	Time Co = R * C: 623.61ns
       -3.45701	Time Co = C * R: 618.93ns
       0.706489	Time Co = C * C: 602.76ns <-- this is the Eigen default!
Timings for dynamic matrix product. R: row major matrix. C: column major matrix. 6x6 matrices.
       0.883127	Time Ro = R * R: 697.61ns
       0.424736	Time Ro = R * C: 704.35ns
       0.650286	Time Ro = C * R: 694.21ns
       0.488761	Time Ro = C * C: 717.8ns
       0.883127	Time Co = R * R: 685.03ns
       0.424736	Time Co = R * C: 691.25ns
       0.650286	Time Co = C * R: 679.14ns
       0.488761	Time Co = C * C: 698.9ns <-- this is the Eigen default!
Timings for static matrix product. R: row major matrix. C: column major matrix. 3x3 matrices.
        2.70709	Time Ro = R * R: 1.86ns
      -0.506792	Time Ro = R * C: 1.87ns
        5.76057	Time Ro = C * R: 1.86ns
       -1.50145	Time Ro = C * C: 1.86ns
        2.70709	Time Co = R * R: 1.87ns
      -0.506792	Time Co = R * C: 1.88ns
        5.76057	Time Co = C * R: 2.01ns
       -1.50145	Time Co = C * C: 1.97ns <-- this is the Eigen default!
Timings for static matrix product. R: row major matrix. C: column major matrix. 4x4 matrices.
       0.227122	Time Ro = R * R: 3.37ns
      -0.855479	Time Ro = R * C: 3.82ns
        2.49703	Time Ro = C * R: 3.19ns
      -0.982448	Time Ro = C * C: 3.24ns
       0.227122	Time Co = R * R: 3.38ns
      -0.855479	Time Co = R * C: 3.82ns
        2.49703	Time Co = C * R: 3.27ns
      -0.982448	Time Co = C * C: 3.01ns <-- this is the Eigen default!
Timings for static matrix product. R: row major matrix. C: column major matrix. 5x5 matrices.
       -1.07814	Time Ro = R * R: 63.79ns
        2.24391	Time Ro = R * C: 69.11ns
     0.00901555	Time Ro = C * R: 62.08ns
       0.776387	Time Ro = C * C: 67.43ns
       -1.07814	Time Co = R * R: 67.45ns
        2.24391	Time Co = R * C: 66.75ns
     0.00901555	Time Co = C * R: 72.66ns
       0.776387	Time Co = C * C: 67.82ns <-- this is the Eigen default!
Timings for static matrix product. R: row major matrix. C: column major matrix. 6x6 matrices.
        2.14536	Time Ro = R * R: 59.32ns
      -0.359196	Time Ro = R * C: 50.66ns
        5.77685	Time Ro = C * R: 63.58ns
       0.293584	Time Ro = C * C: 62.62ns
        2.14536	Time Co = R * R: 65.48ns
      -0.359196	Time Co = R * C: 49.79ns
        5.77685	Time Co = C * R: 61.11ns
       0.293584	Time Co = C * C: 58.94ns <-- this is the Eigen default!
Test q and R rotations
Time w = R * v: 3.3353ns
v norm change: 8.91919e-09 dB
Time w = q * v: 7.2434ns
v norm change: 9.15604e-09 dB
````

AtDinesh, 2016-09-06:
Coming back since we are working on some way not to get rid of the most of Dynamic-sized vectors and matrix.

We should use the EIGEN_MAKE_ALIGNED_OPERATOR_NEW macro.
Please read the 2 links below for more details : 
[https://eigen.tuxfamily.org/dox-devel/group__TopicStructHavingEigenMembers.html](https://eigen.tuxfamily.org/dox-devel/group__TopicStructHavingEigenMembers.html)
[https://eigen.tuxfamily.org/dox-devel/group__TopicFixedSizeVectorizable.html](https://eigen.tuxfamily.org/dox-devel/group__TopicFixedSizeVectorizable.html)

joanvallve, 2016-09-07:
Yes, you are right. We already did it in some classes that use fixed size eigen matrices, [see the commit](https://github.com/IRI-MobileRobotics/wolf/commit/733089cb8fdf2879beefa5abd006587a6ecf3e81)