IRI-MobileRobotics/wolf #91: ProcessorTracker: max_new_features

joanvallve, 2016-08-03:
Currently, this parameter is an unsigned int atribute in `ProcessorTracker`. It is used when it calls the method `detectNewFeatures(max_new_features)` in the derived classes.

Until now, 0 meant unlimited number of new features allowed to be detected.

However, now I realize that there are some cases in which no new features should be detected (localization without mapping, for instance) so 0 value shouldn't mean unlimited.

Should we change it to an integer and use -1 to unlimited or do you think it would be better to stay with an unsigned int and put a huge number when we want unlimited amount of new features?

joansola, 2016-08-03:
My opinion:
- Self-documented code is better. So int with -1: unlimited, 0: no features, n: max features is a good option.
- Otherwise, document very well the parameter and if it appears in constructors and setters, document it there too.

AtDinesh, 2016-08-05:
Otherwise if we wan to keep unsigned int we could use NaN... but I feel like NaNs can be dangerous.