IRI-MobileRobotics/wolf #93: test_processor_imu

AtDinesh, 2016-08-22:
I analysed the results we get with test_processor_imu.
I tested the program with data acquired while the IMU was in a static position.

Here are the first results. In the first figure I just plotted the accelerometer and gyroscope data.. and also the evolution of deltas.
[image]

And we can see that there is something happening at every 0.28 increment in timestamp. If this is obvious on dp_z, zooming let us see that this can also be seen on dp_x and dp_y.

Actually, this is also true for dv. (figure below)
But I can't be sure for dv_y --> variations are so small that I can't see anything.
[image]
But the deltas of quaternion do not seem to have this problem.
maybe the problem comes from some method using the quaternions (v2q etc ?)

I also wanted to oppose the deltas we get to those we would have if we compute them from the delta_v computed in processor_imu : 
[image]
This is likely to be explained by our computation of delta_p in data2delta. We currently use the preintegrated velocity instead of using the delta_v. 

The following figures may not be useful right now but here they are just in case.
Evolution of the state of the system:
[image]

Plotting deltas and preintegrated Deltas:
[image]

dtsbourg, 2016-08-22:
> I also wanted to oppose the deltas we get to those we would have if we compute them from the delta_v computed in processor_imu :

I don't understand what you mean by this ? How did you compute the dp from the delta_v ? Did you just integrate it or change the pre-integrated Delta by the immediate delta ?

dtsbourg, 2016-08-22:
> maybe the problem comes from some method using the quaternions (v2q etc ?)

Could be, quadrant switching perhaps. I will take a look, let you know if I see anything.

dtsbourg, 2016-08-22:
@AtDinesh Can you upload the code and data for the plots somewhere to reproduce ?

joansola, 2016-08-22:
Hello Dinesh, thanks tfor this testing.

I just sent an email to Dylan with info about delta preintegration. Please have a look:

<<<
Bonjour Dylan, how were your holidays? I’ve been a little bit everywhere, and did quite a bit of mountain biking. 

Are you at the lab already? Not me, I’m taking too long this morning, tired from a long road trip yesterday.

In my spare hours I have been working on the Delta preintegration maths. Basically, I have found a way to explain the deltas in an intuitive way, and to express the delta in data2delta in the same format as the other deltas, so that the composition function deltaPlusDelta is valid for both cases:

D+d

D+D

The data2delta is in fact as follows:

dp = 0.5 * q_ij * (a - b_a) * dt^2
dv = q_ij * (a - b_a) * dt
dq = exp ((w - b_w) * dt)

I have prepared a PDF with all the necessary details. Please have a look at it. Especially, read chapter 6, which is a summary of all my findings. I explain the full document here:

Chapter 1 and 2 : intro and notations

Chapter 3 is the case with no gravity. It serves as an easier starter, which works essentially the same. It contains 2 subchapters: first and second order integrations, and both of them work.

Chapter 4 incorporates gravity. It contains 2 subchapters: first and second order integrations. Only the second-order one works.

In WOLF, we have been using first order integration: this will not work. Please see a section in red color in the 1st-order case to detect a term that is not working.

Chapter 5 introduces bias. We know this: just use a-b_a and w-b_w and the rest is the same.

Chapter 6 is a summary of the interesting formulae for the case of interest: gravity and 2nd order. As you can see, the small delta and the big Delta have the same role when it comes to composing them, either additively or subtractively.

Chapter 7 retakes chapter 6 but in a code style, similar to Matlab. Wolf should follow the code here in data2delta and deltaPlusDelta. Also, we can now code deltaMinusDelta, and xx2delta (which could be names predictDelta, or xMinusX).

The thing is: could you please recode data2delta and deltaPlusDelta in Wolf to reflect this? I am talking about the maths, not the API. The API should be OK I think.

I’ll be around late today, or tomorrow. As you may know, this week the lab only opens until 3PM.

Happy return!

I think that we need to have this code right before further analysis. It might be that your comments are related to this.

In fact, by preparing this PDF I realized that, I think, the Delta_p in the Forster’s paper (eq 33) is wrong! 

Joan

AtDinesh, 2016-08-22:
@dtsbourg 
What I mean is that I just wanted to see what I would get if I just use the following equations (That I think we should be using). 
`dp = (3/2)*dv*dt`

instead of : 
`p_out_ = velocity_preint_ * _dt;`

that would be : 
`dp = Dv * dt;`

joansola, 2016-08-22:
I think we must refer to this document for proper integration:

[IMU-preintegration.pdf](https://github.com/IRI-MobileRobotics/wolf/files/429808/IMU-preintegration.pdf)

Equation 33 in Forster's paper seems to be wrong with regards to Delta_p.

AtDinesh, 2016-08-22:
In case you still want the files : 
https://1drv.ms/f/s!AslKcxNJfxFBjp4cRvbWQ0wWVfAfVQ

debug_results.dat is the file created by test_processor_imu
the other .dat files should be those I used to run the test
the debug_check.m file is the matlab file to load data and plot results

dtsbourg, 2016-08-22:
Perfect thanks !

joansola, 2016-08-24:
Forster just uploaded a version of his paper with the bug on eq. 33 corrected, [check it out here](https://arxiv.org/pdf/1512.02363.pdf)

dtsbourg, 2016-08-26:
Hey I have a question, maybe one of you can tell me what's going on (or confirm a mistake 😉)

I the paper [1] they use the following expression to compute the jacobian of R wrt the gyro bias (p16, top-right corner, no equation number) : 
[image]

Yet in the code, it seems they are subtracting ```J_r * dt``` instead of multiplying ... 

* [GTSAM/gtsam/navigation/IMUFactor.h -- l.183](https://github.com/devbharat/gtsam/blob/master/gtsam/navigation/ImuFactor.h#L183)
* [GTSAM4/gtsam/navigation/ManifoldPreintegration.cpp -- l.99](https://github.com/haidai/gtsam/blob/master/gtsam/navigation/ManifoldPreintegration.cpp#L99)

Any ideas ? 

[1] https://arxiv.org/pdf/1512.02363.pdf

joansola, 2016-08-26:
They might be using the Linear Jacobians, therefore escaping the SO(3) manifold, and they might just accept it as a convenient approximation? After all, IMUs are integrated at very hight rates, so Dt is very small.
I can't tell you more... :-(

AtDinesh, 2016-08-29:
Hi there !
**ISSUE 1 :**
Concerning -jacobian1 and _jacobian2 in deltaPlusDelta. On friday we said that these should be the jacobians A and B in FORSTER, i.e. jacobians related to noise.
However, the computation of these jacobians implies the use  of _data (at least for A).
**SOLUTION**
As a result, I think that these two jacobians should be the jacobians related to bias that we need to update. For those related to noise, the simplest way would be (I think...) to compute them in data2delta().

**ISSUE 2**
Because of the use of _data, it would not make sense to compute the jacobians related to noise during the composition of 2 preintegrated deltas (with the current formula). 
So, even if the integration of a preintegrated Delta and a delta is similar to the Deltas composition, we may need to create a new function in this last case to perform the correct operations on jacobians.
A new operation like "let's do the Markov assumption and suppose that the noise does not depend from past so, if I have 2 preintegrated Deltas (first one from i to j, second one from j to k), then the jacobian of noise from i to k is the one from j to k " ?

_Making things clear in my head I'll come back when I'll have a better solution_

joansola, 2016-08-29:
__ISSUE 1__
Let us suppose deltaPlusDelta() is such that
D3 = D1 + D2;
then:
- _jacobian1 should be the Jacobian of D3 wrt D1, that is, 

    D3_d1 = d D3 / d D1
- _jacobian2 should be the Jacobian of D3 wrt D2, that is, 

    D3_d2 = d D3 / d D2
- We should compute the covariance of D3 as

    D3_cov = D3_d1 * D1_cov * D3_d1' + D3_d2 * D2_cov * D3_d2'

This is so, because the base class ProcessorMotion is using them to compose noise. We cannot change these definitions without altering the base class.

__ISSUE 2__
Let us suppose now that we start from _data, in such a way as:
D2 = data2delta(_data,dt);
D3 = D1 + D2; // using deltaPlusDelta()
then:
- We need the Jacobian of data2delta wrt data:

D2_d = d D1 / d _data
- We should compute the covariance of D2 as

D2_cov = D_d * data_cov * D2_d'
- We should compute the covariance of D3 as

D3_cov = D3_d1 * D1_cov * D3_d1' + D3_d2 * D_d * data_cov * D2_d' * D3_d2'

this implies that, with respect to Forster's paper:

A = D3_d1
B = D3_d2 * D2_data

joansola, 2016-08-29:
My second answer refers to the fact that the covariances must be in minimal space (9x9-matrices), whereas the deltas are in non-minimal space (10-vectors). This implies little modifications, which are taken care of in Forster paper. However, the APIs in data2delta() and deltaPlusDelta() should not be altered beyond these size issues.

joansola, 2016-08-29:
One last comment: We could actually redefine the base class so that there is no need for data2delta(), and all the work is done in a single function called integrate():

Delta = integrate(Delta, data, dt)
{
    Delta = Delta (+) [1/2 * a * dt^2 ; a * dt ; exp(w * dt) ]
}

However, WOLF still makes use, potentially, of a function called deltaPlusDelta(), for which we need to be able to compute the Jacobians, and to propagate with them the covariances. This deltaPlusDelta() is thought to compose two deltas of any type (e.g. two consecutive pre-integrated deltas). My pdf shows that deltaPlusDelta(Delta_preint, Delta_preint) and deltaPlusDelta(Delta_preint, delta) are the same function.

This is the reason why I decided to use a data2delta() anyway, and reuse deltaPlusDelta() for both preintegration and delta composition.

But we could, I insist, use a single function for the preintegration, if this eases up the construction of the Jacobians A and B instead of the ones in my post above.

AtDinesh, 2016-08-30:
OK so this makes things clear about _jacobian1 and _jacobian2.
Then we can just define static size matrix to store the jacobians wrt noises and biases.

For the computation of A and B,
I did not see this answer before so I have been looking for some solution. Right now I have been facing mainly the problem of D (+) d --> meaning the preintegration at every time increment.
So one solution is to compute A inside data2delta() function.
Computation of B can be done in the integrateDelta() function, just after deltaPlusDelta(). 
From A and B and with a model of noise we can compute the new noise vector.
We also can compute the covariances.
==> This I think should resolve the first 2 points of issue 2.

joansola, 2016-08-30:
No no, it's much simpler!
- Jacobian `A` is exactly `_jacobian1`, and it is computed in `deltaPlusDelta()`
- Jacobian `B` is the product of `_jacobian2` times the Jacobian of `data2delta()`. It is not needed, because the idea is to __proceed as follows__:

1. in `data2delta()`, compute `delta_` and its Jacobian, `_jacobian_delta`, which is 9x6
2. in `data2dellta()`, compute the covariance of `delta_`, as
`delta_cov = _jacobian_delta * data_cov * _jacobian_delta';`. This can be done in the base class ProcessorMotion, but it is currently done in the derived class -- See Note (1) at the end.
3. in deltaPlusDelta(), compute Delta_, _jacobian1 and _jacobian2, which are 9x9
4. After deltaPlusDelta(), compute the covariance of Delta_, as
`Delta_cov = _jacobian1 * Delta_cov * _jacobian1 + _jacobian2 * delta_cov * _jacobian2;`. This can be done in the base class ProcessorMotion.

Thus, to summarize and make it __general for all processors__, the idea is very simple:
1. In the base class:
    - call the functions `data2delta()` and `deltaPlusDelta()`, and obtain all Jacobians
    - perform covariance propagation according to the formulas above
2. In the derived class:
    - implement `data2delta()`, with Jacobians
    - implement `deltaPlusDelta()`, with and without Jacobians

And it is precisely this generalization what obliges us to proceed this way: just implement the functions and compute their respective Jacobians. Nothing else: the rest is taken care of in the base class.

You can look at these functions in `ProcessorOdom2D.h`. There, you will see that:
1. in fact, `data2delta()` computes the `delta_`, the Jacobian, and finally the covariance `delta_cov_`, which is a member of the class. See note (1) at the end.
2. On the contrary, `deltaPlusDelta()` only returns `Delta_` and the two Jacobians, and the covariance propagation is performed by the base class.

Here are the data members present in `ProcessorMotion.h`:

`        Eigen::VectorXs delta_;                 ///< current delta`

`        Eigen::MatrixXs delta_cov_;             ///< current delta covariance`

`        Eigen::VectorXs delta_integrated_;      ///< integrated delta`

`        Eigen::MatrixXs delta_integrated_cov_;  ///< integrated delta covariance`

`        Eigen::VectorXs data_;                  ///< current data`

`        Eigen::MatrixXs jacobian_prev_;         ///< jacobian of delta composition w.r.t previous delta integrated`

`        Eigen::MatrixXs jacobian_curr_;         ///< jacobian of delta composition w.r.t current delta`

where `jacobian_prev_` is `_jacobian1` and `jacobian_curr_` is `_jacobian2`. I would actually rename _jacobian1 and 2 so that they match the member names, but this is secondary.

So, the best thing is to follow the logic of `ProcessorOdom2D`.

As I said in another post, there is this second issue of the Jacobians wrt biases. This must be taken care of in `deltaPlusDelta()` because the base class has nothing to do with these Jacobians (in fact it ignores their existence completely).

__Note (1)__: The fact that `data2delta()` computes the covariance is indeed odd regarding the generalization and the roles of the base class and the derived class. I can change it so that it is the base class who is computing the covariance, leaving the derived class with the exclusive job of computing the `delta_` and the Jacobians.

joansola, 2016-08-30:
Please see the code in ProcessorMotion::integrate():

```c
inline void ProcessorMotion::integrate()
{

    // Set dt
    updateDt();

    // get data and convert it to delta
    data2delta(incoming_ptr_->getData(), incoming_ptr_->getDataCovariance(), dt_);

    // then integrate the current delta to pre-integrated measurements
    integrateDelta();

    // and covariance
    deltaCovPlusDeltaCov(getBufferPtr()->get().back().delta_integr_cov_,
                         delta_cov_,
                         dt_,
                         jacobian_prev_,
                         jacobian_curr_,
                         delta_integrated_cov_);

    // then push it into buffer
    getBufferPtr()->get().push_back(Motion( {incoming_ptr_->getTimeStamp(),
                                             delta_,
                                             delta_integrated_,
                                             delta_cov_,
                                             delta_integrated_cov_,
                                             Eigen::MatrixXs::Zero(delta_size_, delta_size_),
                                             Eigen::MatrixXs::Zero(delta_size_, delta_size_)}));
}
```
and recall that `deltaCovPlusDeltaCov()` is a function of the base class `ProcessorMotion`, not of the derived class. See also that `delta_cov_` is needed, because it is pushed into the buffer.

dtsbourg, 2016-09-02:
I was stuck on some type conversion stuff while working on the constraint, which is when I found this in the [Ceres FAQ](http://ceres-solver.org/faqs.html) : 

> When using Quaternions, consider using QuaternionParameterization.
>
> Quaternions are a four dimensional parameterization of the space of three dimensional rotations SO(3)SO(3). However, the SO(3)SO(3) is a three dimensional set, and so is the tangent space of a Quaternion. Therefore, it is sometimes (not always) benefecial to associate a local parameterization with parameter blocks representing a Quaternion. Assuming that the order of entries in your parameter block is w,x,y,z, you can use QuaternionParameterization.
>
>If however, you are using Eigen’s Quaternion object, whose layout is x,y,z,w then we recommend you use Lloyd Hughes’s Ceres Extensions.

And indeed, Ceres doesn't like Eigen's quaternions. So do you think we should : 
* [ ] Include the [ceres_extensions](https://github.com/system123/ceres_extensions)
* [ ] Switch to rotation matrix all the way through the constraint creation process (which kind of defeats the purpose of using quaternions in the first place)
* [ ] Find a way to use Ceres' [Quaternion Parametrization](http://ceres-solver.org/modeling.html#QuaternionParameterization)

joansola, 2016-09-02:
Hi Dylan,

The class StateQuaternion derives from StateBlock and adds a LocalParametrizationQuaternion, exactly by following Ceres guidelines. You should not worry about this.

In FrameIMU, you should see that the orientation block is a StateQuaternion (is it? see constructor in FrameIMU line 34)

I may need to revise if the constructors above, which accept all StateBlock, should instead require StateQuaternion in their API.

J

sandwichmaker, 2016-09-06:
Ceres now has EigenQuaternionParameterization 
https://github.com/ceres-solver/ceres-solver/commit/d4264ec10d9a270b53b5db86c0245ae8cbd2cf18
adds it.

joansola, 2016-09-06:
Thanks @sandwichmaker ! We already coded it, but it's nice that ceres finally added this! By the way, how did you notice about our issue?

sandwichmaker, 2016-09-06:
@joansola I look around at people using and sometimes complaining about ceres and work to alleviate their misery :)

joansola, 2016-09-06:
Just FYI: we do not use Ceres directly in our project but just as an option. We therefore built our LocalParametrizationQuaternion, and then have a wrapper module bridging it (and a bunch of other things) onto Ceres in case we use Ceres. This wrapper now uses the regular ceres::LocalParametrization because we already solved the algebra for the quaternion in our class.
And another 'by the way'> the algebra for a 3D homogeneous vector local parametrization is exactly the same as for an Eigen-based quaternion, as they both constrain their parameters to the unit sphere in R^4, and have their elements ordered in the same way (x,y,z,w).
Thanks a lot anyway.

AtDinesh, 2016-09-08:
test_processor_imu stopped working some time ago. here is the problem : 
We give processor_motion Size parameters to be able to set some matrices. Here, for the jacobian matrix w.r.t. deltas we define it as beeing of size (delta_size_, delta_size). that makes sense in the general case and for processor_odom ! However, in the IMU case, the deltas are of size 10 [(P,V,Q), with Q containig 4 parameters]. But we decided to represent jacobians and covariance matrices in minimal form ! that is 9x9 ! So here is the problem --> the covariance computation method tries to compute (9x9) * (10x10).

Simplest solution would be to add another parameter in the constructor of Processor_Motion to get the actual size of jacobians. I am working on it. If you have any other solution to propose...

joansola, 2016-09-09:
Waiting for confirmation: is processor_imu computing all the covariances yet? If so, these are the performance figures in my PC:
```
Statistics -----------------------------------------------------------------------------------
If you want meaningful CPU metrics, remove all couts in the loop / remove DEBUG_RESULTS definition variable, and compile in RELEASE mode!
t0        : 418.535 s
tf        : 434.729 s
duration  : 16.194 s
N samples : 16201
frequency : 1000.364 Hz
CPU time  : 0.146 s
s/integr  : 9.015 us
integr/s  : 110922.436 ips  <-- 110X faster than IMU frame rate
```
which is really good! I wonder all the computations are not there yet...

AtDinesh, 2016-09-11:
Are the covariances wrt residuals computed by Constraints ?

joansola, 2016-09-15:
@AtDinesh you mean the Jacobians? Yes, automatically using ceres::Jet.