IRI-MobileRobotics/wolf #96: Deltas in ProcessorMotion, and especially IMU and its constraints

joansola, 2016-09-06:
These are the things that I want to do:

1. Ensure all Deltas, Jacobians and Covariances are ordered P V Q. This is the order in which we need to perform operations in functions to avoid aliasing. Currently, we have:

  - Deltas in processor_imu are PQV

  - Jacobians wrt bias are PV and PVQ

  - Covariances are PVQ I think

2. Make the Deltas a struct and not a VectorXs. This would eliminate part of the point above, but not all since Covariances would continue to be compact matrices 9x9, and the same for some of the Jacobians. This is a big refactoring, since it affects ProcessorMotion and therefore ProcessorOdom2D and ProcessorOdom3D, and also CaptureMotion.

3. In ConstraintIMU, I think the best thing is to have all chunks cut into pieces, like dp1, dv1, dq1, dp2, dv2, dq2, dab1, dwb1, and the Jacobians dDp_dab, dDv_dab, dDp_dwb, dDv_dwb, dDq_dwb.

4. In ConstraintIMU, and possibly also in ProcessorIMU, have all helper functions be dependent either on the Delta structs, or on the Delta chunks.

5. Have a file called imuTools.h where all these functions, plus all exp and log, plus skew and vee, are located.

I think all this work is needed if favor of clarity and rationality. Also, I am convinced that, the sooner we do it, the sooner we will be able to proceed quickly, as now we are in a quite confusing state.

What do you think?

joansola, 2016-09-06:
__Dinesh response__

Talking about the this rationalization, this would mean that we create a struct Delta in processorMotion. This struct would contain other containers like vectors, matrix, etc,.. that are common to all the processors derived from ProcessorMotion. Methods like deltaPlusDelta will be modified to use these Delta struts.

In classes derived from processorMotion, for example in processor_imu, we can define another struct derived (struct Delta_imu : public Delta) from Delta and containing more parameters/methods in static size. And we would do the same in other classes like processor_odom.
So this should be OK !

Actually this is like using struct like any other class in C++. The main difference with C++ here would be that default inheritance of struct is public.

Do we agree up to this point ?

joansola, 2016-09-06:
Yes, there were issues with templates, I cant remember of what kind...

joansola, 2016-09-06:
For example :
```
struct Delta{
    Matrix<Scalar,3,1> p, v;
    Quaternion<Scalar> q;
}
```
and also:
```
Matrix<Scalar,3,3> dDp_dab, dDv_dab, dDp_dwb, dDv_dwb, dDq_dwb;
Matrix<Scalar,9,0> delta_cov; 
```

joansola, 2016-09-06:
Having DeltaBase and DeltaIMU : public DeltaBase is a way to go. The APIS should then work with DeltaBase*, and use polymorphism as usual.

Another option is having ProcessorMotion, CaptureMotion template classes of a certain Delta. Then, in processor_motion.h, we define DeltaIMU, and do ProcessorIMU : public ProcessorMotion<DeltaIMU>.

joansola, 2016-09-06:
__Dinesh response_

But with the second option are we not going to face the same problems we already faced some time ago and due to what we though that using templates in processorMotion is a bad idea?

I can't remember exactly the problems that we had.

joansola, 2016-09-06:
Yes... the thing is that I can't remember either what was wrong with the template Deltas...

What I don't quite like from the dynamic polymorphism based on pointers is that we need to keep doing these new and delete... but it's maybe the way to go.

joansola, 2016-09-06:
__Dinesh response__

I have been looking for problems related to the use of new / delete. No one seem to agree on the internet.
I will think about some solutions to avoid to abuse of new() and delete[]..

joansola, 2016-09-06:
__Dinesh response__

I have been thinking about a solution to avoid the use of the new and delete if we want to create DeltaBase, DeltaIMU, etc..

I am not sure that this is the right thing to do, but if we take the example of processor IMU here, it could create a container ( map, deque, vector, we would have to choose the most efficient one for our application). This container can then store pointers to Delta objects. However, if these are normal pointers it might not work as expected but if these are smart pointers (unique_ptr, shared_ptr..) for example, then deleting old Deltas can be done automatically (from what I understood).

Now let's consider this :

We have ptr_container, a container containing pointers to Deltas,
Then we have all the Deltas from last keyframe to current frame.
Now if we finally decide to make of the current frame a KeyFrame, this means we want to use the preintegrated Delta to make an IMU_Factor. This factor already contains all the information that we need so all that is contained in ptr_container should be deleted. This can be done be resetting all the pointers contained in this container.

This might be a naive way to go and certainly not the best one..

Another alternative would be to use pools : http://gameprogrammingpatterns.com/object-pool.html

joansola, 2016-09-06:
Oh I see. In problem, there is  a pointer to ProcessorMotion. If processorMotion is template, then Problem might become templated too, and this we cannot afford.

joansola, 2016-09-06:
Yes, what you say is correct: a container of pointers to ```DeltaBase```, then use polymorphism to cast to ```DeltaIMU```. If these pointers are smart, then all works fine. Otherwise, we need to face the ```new``` / `delete` issue. With smart we still have to face the new anyway.

Moreover, if we create a class `DeltaBase` and we make the following methods pure virtual:

```
Zero()
Inverse()
Plus(DeltaBase* )
Minus(DeltaBase*)
Predict(VectorXs & State1* , VectorXs & State2*)
Minimal()
```

then all the calls could be done in ```ProcessorMotion```.

However, I think this is an overkill, as all these issue with Deltas is only interesting for IMU.

joansola, 2016-09-06:
How about this. Use DeltaBase and DeltaIMU class with:

```
struct DeltaBase{ 
    TimeStamp t_final;
    Scalar dt;
    virtual setZero() = 0;
} 

struct DeltaIMU : public Delta{
    Vector3s p;
    Vector3s v;
    Quaternions o;
    setZero();
}
```

joansola, 2016-09-06:
We can add these extra methods to the base class (always pure virtual)

```
struct DeltaBase{ 
    TimeStamp t_final;
    Scalar dt;
    virtual setZero() = 0;
    virtual Inverse() = 0;
    virtual plus(DeltaBase* ) = 0; // or better overload operator +
    virtual minus() = 0; // or better overload operator -
    virtual setFromStates(VectorXs x1, VectorXs x2, TimeStamp t1, TimeStamp t2)
} 
```

but as I said, I feel this is an overkill

AtDinesh, 2016-09-06:
Note for myself : 
Creation of Deltas :

AtDinesh, 2016-09-06:
Note for myself : 
Creation of Deltas occurs : 
   - in data2delta
   - in deltaPlusDelta (can be avoided)
   - in deltaMinusDelta

deleting Deltas : 
  - At Keyframe / Factor generation

nmansard, 2016-09-06:
Are you sure that Deltas would be used by functions that donot know their exact type?

AtDinesh, 2016-09-06:
One way to avoid using `new` would be the use of allocators. But I don't know much about these.
The STL provides [std::alocator](http://en.cppreference.com/w/cpp/memory/allocator). But this default allocator uses the operator `new` to allocate memory. ([more details here](https://en.wikipedia.org/wiki/Allocator_(C%2B%2B))). 

here is a quote from this last link : 
> One of the main reasons for writing a custom allocator is performance. **Utilizing a specialized custom allocator may substantially improve the performance or memory usage, or both, of the program**.[4][15] The **default allocator uses operator new** to allocate memory.[16] This is often implemented as a thin layer around the C heap allocation functions,[17] which are usually optimized for infrequent allocation of large memory blocks. **This approach may work well with containers that mostly allocate large chunks of memory, like vector and deque.[15] However, for containers that require frequent allocations of small objects, such as map and list, using the default allocator is generally slow.**[4][17] Other common problems with a malloc-based allocator include poor locality of reference,[4] and excessive memory fragmentation.[4][17]

> **A popular approach to improve performance is to create a memory pool-based allocator.[15] Instead of allocating memory every time an item is inserted or removed from a container, a large block of memory (the memory pool) is allocated beforehand, possibly at the startup of the program.** The custom allocator will serve individual allocation requests by simply returning a pointer to memory from the pool. Actual deallocation of memory can be deferred until the lifetime of the memory pool ends. An example of memory pool-based allocators can be found in the Boost C++ Libraries.[15]

Here is another link to understand what a memory pool is. ([http://www.codeproject.com/Articles/27487/Why-to-use-memory-pool-and-how-to-implement-it](http://www.codeproject.com/Articles/27487/Why-to-use-memory-pool-and-how-to-implement-it))

I feel like the method that a proposed with containers is acting like a simple version of memory pool. 
I don't know if the use of a memory pool is justified in our case. But for now we can create a simple pool as a container of pointers to DeltaBase in processorMotion.

AtDinesh, 2016-09-06:
deltas should be used in processorMotion and its derived classes (imu, odom, etc..), keeping in mind that processorMotion is an abstract class.
At some point, the processor creates a constraint (which is a factor) using those Deltas. Then I think that all the operations will be done directly from this constraints (using the base class ConstraintBase). So no, I think the Deltas type should be known by functions using it.
But I would wait for @joansola 's answer to confirm this.

joansola, 2016-09-06:
Oh, this pool thing is really cool!
I'll have some more time to write my ideas later, now I'm on other issues. Thanks for the contributions!
I want to keep simple though. The thing now is that, for trying to keep simple, we ended up complicating our lives. This is common, as the say goes, 'the easiest thing in life is to make your life complicated'

joansola, 2016-09-06:
But quickly: I see the pool thing as follows:
 - We have a buffer called MotionBuffer containing a std::map structs called Motion where the key of the map is a TimeStamp. A Motion object contains a delta, a delta_preintegrated, some Jacobians, some covariances matrices, and some time stamps.
 - This buffer grows at the IMU frequency (say 1kHz)
 - a pool for the buffer would be OK to store Motion objects, not pointers to them, otherwise there is no benefit.
 - We can build this 'pool' very easily with just a std::vector with a preallocated size sufficiently large. Or with a std::deque.
 - Also, before storing into the pool, we should be clear to ensure that we are reusing a single instance of a Motion object, or better a pointer to one empty instance of Motion in the pool, so that we don't go creating / copying / deleting temporary Motion objects anyway.

AtDinesh, 2016-09-06:
So we can either keep the MotionBuffer or replace it with a 'pool'...

If we want to build the pool from a std::vector or std::deque, let's keep this in mind : 

**std::vector** is a dynamic-array class.  Useful when the size of the data is known and is known to not change too often. It is also good when you want to have random-access to elements. in our case, We could create a very large vector (10K elements for example, corresponding to ~10 secs with IMU). But this means that some part of the memory would not be used in most cases / and we cannot make sure that we will never reach those 10K, if we reach it then we will have to face reallocation that will invalidate all the references, pointers, and iterators referring to the elements in the sequence.

**std::deque** is a double-ended queue that can act as a stack or queue. Good for when you are not sure about the number of elements and when accessing data-element are always in a serial manner. They are fast when elements are added/removed from front and end but not when they're added/removed to/from the middle. Pushing and poping methods at one of the end of the queue are O(1). Reallocation occurs in fewer cases but references are invalidated when inserting / erasing in the middle of deque. `pop_front()` and `pop-back()` only invalidate iterators/references to the element at the end in question. So `push_front()` and `push_back()` will invalidate iterators, but references to the elements themselves remain valid.

There are cons and pros for both.
But we should not forget that both vector and deque have a maximum size (that may not be the same in both cases)! 

[http://www.cplusplus.com/reference/deque/deque/max_size/](http://www.cplusplus.com/reference/deque/deque/max_size/)
> This is the maximum potential size the container can reach due to known system or library implementation limitations, but the container is by no means guaranteed to be able to reach that size: it can still fail to allocate storage at any point before that size is reached.

This is true for both deque and vector !

joansola, 2016-09-06:
A response to Nico: ProcessorMotion is a base class for 3 processors already: ProcessorIMU, processorOdom2D, and processorOdom3D.
All of them manipulate deltas in the same way, with a bufffer, etc. But the deltas in all of them are different.
The methods in ProcessorMotion therefore don't know the exact type of delta.
The solution we took initially has been to define all deltas as VectorXs, with dynamic size. Now we are discussing on how to make it better:
 - templates
 - polymorphism
 - void methods in ProcessorMotion
The rest of the discussions here are just side-topics not directly related to the main subject

joansola, 2016-09-15:
OK, finally a word about this.

This issue with the Deltas is nearly a __no-issue__. We'll keep the VectorXs form, since CPU performance is showing very good. Actually, when we consider the amount of operations in a single Delta integration, the part involving Jacobians and covariances is by far the most costly one. then, having the initial variables in a dynamic-size VectorXs makes no real difference.

We did actually update the IMU deltas order to be PVQ instead of PQV.