IRI-MobileRobotics/wolf #97: Migration to GitLab

joansola, 2016-09-16:
Hello, @AtDinesh , @JaimeTarraso , @artivis , @andreucm , @joanvallve , @joansola

We are migrating WOLF to GitLab. We will do it on Monday 19/09/2016 at 10AM. In the meantime:
 - Please acknowledge reception of this communication.
 - Please indicate if you have any inconvenients with this migration.
 - Please proceed with any pending commits and push to GitHub.
 - Please sign up to IRI-GitLab using your GitHub account: go to https://gitlab.iri.upc.edu/users/sign_in, click on 'sign in with github' or 'sign in with gitbucket' (preferrably github)
 - Please tell me when you have a GitLab accout, and provide me with the username. 
 - If you already have a GitLab account, use that one if you wish. Please tell me so so that I can properly manage your permissions.

On my side, I am taking care to provide access to IRI-GitLab to your accounts.

Thank you!

@nmansard , you are on copy of this message, do as you wish. From our side, we are content to have you in the loop.

artivis, 2016-09-16:
@joansola Got it ! No inconvenient.
I tried to login with github but it says that my account is blocked. I guess I have to wait for a permission.

joansola, 2016-09-16:
Yes you will need permission.

AtDinesh, 2016-09-19:
OK for me !
Same for me my account is blocked.

joansola, 2016-09-19:
Usually you do not have permissions to access IRI-GitLab from GitHub.

I will therefore ask for permissions for your GitHub account.

I'll keep you posted.

Joan